bitfocus/companion-module-birddog-ptz #2: Preset Drive Speed

jgwhitebeard, 2020-11-01:
Thanks for your work on this. Really pleased to see the module added.
Have been playing with the test build on my P200s and can't seem to make Preset Drive Speed work. I've tried it on it's own and it does nothing. If I try recalling the referenced preset after using Preset Drive Speed to set it to speed 1 (slow) it just moves at full speed. Am I missing something? Same happens when using Sony Visca.
Thanks.

BenniMTools, 2020-11-05:
I will take a look into it tomorrow.

BenniMTools, 2020-11-07:
Hi,
it looks like the Birddog Camera don't support to change the Preset Drive Speed.
You can't even change it in the Birddog Camera Control.
I removed the command.
If you find a command to change the speed let me know and i will add it.

closed 2020-11-07 by BenniMTools