bitfocus/companion-module-birddog-ptz #4: Recall Preset Problem

keester03, 2020-12-28:
When I set my Recall Preset command it seems to recall the next preset number instead of the one I set. Ex: when I recall Preset 1 the camera moves to the Preset 2 location, and so on....Preset 8 triggers Preset 9, etc.

BasilChessman, 2020-12-28:
It sounds that perhaps the preset id starts at zero instead of one.

keester03, 2020-12-28:
It does seem to act like this. How could this get tested/fixed?

BenniMTools, 2020-12-29:
What camera and what firmware are you using ?

keester03, 2020-12-29:
Using Birddog P200 cameras with the latest baseboard firmware and the recently available beta camera firmware.

BenniMTools, 2020-12-29:
it looks that BD changed the preset numbers.
I Programmed it with the 3.3 and there everything was right.
I take a look that i make a checkbox if you use version > 3.3 to be compatible for the next weeks until they have a stable release version from the firmeware

keester03, 2020-12-30:
Thanks BenniM, I’m a bit unsure of what you mean, did you say you will revise your code when a stable released FW version is launched by BD, perhaps in the next few weeks?

BenniMTools, 2021-01-23:
Fixed. Thanks.

closed 2021-01-23 by BenniMTools