bitfocus/companion-module-birddog-ptz #5: Birddog P200

sellbud, 2021-01-29:
There are many useful commands for the P200, but I was hoping that I could have buttons to increase and decrease the contrast. In the web interface it is designated as contrast, but on the PTZ keyboard it is described as Black level. I can't find a list of the Birddog Visca commands anywhere to experiment on my own.

BenniMTools, 2021-01-29:
hey sellbud,
since version A2_A3 in the biddog camera it should be possible.
I got the new command list a few days ago and i don't had a camera to test the commands thats the reason why it's not in.

unfortunately there is no command to increase or decrease the contrast. but you can send a value.
 the command should be 81 01 05 0d 01 0p 0q FF where pq should be 0-15.

if you can test it, it would be nice to hear from you.
Best,
Benni

bryce-seifert, 2022-05-02:
Contrast is now available via https://github.com/bitfocus/companion-module-birddog-ptz/commit/d9a1bf3966365ede85a72d7995cbe7d06418e312 for cameras listed as supporting this feature by BirdDog (BirdDog P100 | BirdDog PF120)

closed 2022-05-02 by bryce-seifert