bitfocus/companion-module-birddog-ptz #6: P200 Feedback

smalljoshua, 2021-02-19:
Hey Folks,

I'm running companion 2.1.2 with the birddog visca module, cameras are latest firmware (21.02.3384).

I've tested the following:
I've tested the majority of the functions and everything appears to work correctly except the feedback for any of the settings:
Gain
Shutter
Iris
Red
Blue

I can see the information is being correctly sent by the camera to the machine running companion as verified in wireshark against the datasheet from BirdDog

https://s3.amazonaws.com/cdn.freshdesk.com/data/helpdesk/attachments/production/48069665179/original/Visca%20command%20list_new.PDF?response-content-type=application%2Fpdf&X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAS6FNSMY2WD6T3JNC%2F20210219%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210219T180508Z&X-Amz-Expires=300&X-Amz-SignedHeaders=host&X-Amz-Signature=8ef92a60a5b68a3e532bdeef69127f56e2ebf595c51aef6fcae1ebcd6eae82dc

It would also be good to get the current exposure status available as feedback if possible to tell when we're in manual vs automatic etc.

BenniMTools, 2021-02-20:
Hi,

the problem is that the feedback functions are for the old firmware and the new is different.
Unfortunately i don't have a new camera with latest firmware here so i can't test the new commands. :-(

smalljoshua, 2021-02-20:
Hey Benni, if you're not too far (southwest UK) I can lend you a P200 to play with. Alternatively, happy to be a guinipig (I've not got time to setup compile though hence needing some assistance)

BenniMTools, 2021-02-22:
Hey,
i live in Germany. But I'm the next month in Switzerland.
I will try to talk to a friend, maybe he have one for me the next weekend.

dominik34, 2021-04-07:
Hi, i've got a similar problem with the Gain, Exposure, Shutter,... Feedback. I'm using several P200s with the latest firmware. Sometimes the Feedback works, sometimes not.

If I could help testing new commands for the latest firmware, do not hesitate to contact me.
I love the opportunity to control my cams with companion and would be happy to help improving this module...

dominik34, 2021-04-07:
Okay, yes I will test it if i can.

By the way I found something out which could eventually help you while correcting this issues with the feedback.
First I was wondering about the different values in companion in comparison to them on the BirdDog web interface:
[image]
Than I put red up to 255 in companion but on the webinterface it was only at 170. After changing the Red-Value in the webinterface the companion button got stuck and i was't able to change anything here until I switched to another wb-mode and back.
At least I can't yet see any regularity between the different numbers.
But if something gets changed in the webinterface, the feedback in companion stops working.

PS: Wegen mir können wir uns auch auf Deutsch unterhalten ;-)

BenniMTools, 2021-04-07:
Hey,
Ok cool 😝.
Jo damals gab es diese correction bei den cams noch nicht.
Ich hab die Werte mit den Werten im osd verglichen.
Aber gut zu wissen. Dann weiß ich schonmal wo ich suchen kann.
Wo kommst du den her ?

dominik34, 2021-04-08:
Baden-Württemberg - südlich von Ulm ;-)

wikipeter-de, 2021-04-19:
> PS: Wegen mir können wir uns auch auf Deutsch unterhalten ;-)

Wow, du bist ja schon ganz ordentlich mit Companion und den Birddogs unterwegs. Ich bin gerade auch dabei eine ähnliche Lösung zu basteln. Bei uns in der Kirche kommt bei fortschreitender Helligkeit bei den Abendgottesdiensten die Sonne rein. Da musste ich jetzt die Blende mal korrigieren. Wir fahren normalerweise mit "Shutter Priority". Dafür habe ich einen Button mit Exposure "Manual" und "IRIS DOWN" belegt. Das klappt ja wunderbar. Aber interssieren würde es mich, wie man an die Blendenwerte kommt um diese auf einem Button anzuzeigen.

Und hast du schon was mit der Restful API von Birddog gemacht? Habe da noch nichts über Companion hinbekommen. Wollte da die Blendenwerte auslesen. 

Mit welchen Einstellungen betreibt ihr eure Kameras (P200?) in der Kirche? Wir so ... https://www.youtube.com/watch?v=qpJPexp8zsU

dominik34, 2021-04-20:
Es gibt bei den Presets einen Block "Feedback" dort liegt auch ein Button für exposure feedback drin. (Heißt eventuell anderst) einfach mal alle ausprobieren dann siehst du schon welcher es ist 😉

mit der Restful API hab ich bisher nichts gemacht. Allerdings gibt es auf: https://desk.zoho.com/portal/api/kbArticles/574085000000642627/attachments/6r81le95f4132358f4c51bd67954c5ff79410/content?portalId=edbsnfaffdf840d1eef8c81ee4ee9dfacd8a48be54e0a2e0cd1acde19c38c2612ea9e&inline=true
Ein PDF mit den ViscIP Befehlen für die P200. Du kannst also in Companion einen Custom command einfügen und dort jeden beliebigen visca Befehl eintragen.
(Bei der Schreibweise muss allerdings noch etwas angepasst werden.)

wir haben zwei P200 im Einsatz und ich arbeite meistens mit Iris oder Shutterpriority. Bin momentan aber selber noch am herumprobieren was am besten ist.

PS: so sieht es bei uns aus 😉 https://youtu.be/mg3Va-_sOqQ

wikipeter-de, 2021-04-20:
> Es gibt bei den Presets einen Block "Feedback" dort liegt auch ein Button für exposure feedback drin. (Heißt eventuell anderst) einfach mal alle ausprobieren dann siehst du schon welcher es ist 😉
> 
> PS: so sieht es bei uns aus 😉 https://youtu.be/mg3Va-_sOqQ

Ah, den Block "Feedback" bei den Presets habe ich dann bisher irgendwie übersehen. Das checke ich mal.
Ihr habt ein ordentliches Bild. Wie weit bist du mit der Kamera von hinten weg? Bei uns sind es glatte 25 Meter bis zum Altar. Das macht natürlich bei abendlicher Dunkelheit und nicht optimaler Beleuchtung was aus. Wir haben dann immer mal wieder ein ziemlich schlechtes Bild mit Artefkaten rund um den der sich da gerade bewegt. Bei Twitch ist es aber besser. Da streamen wir parallel einfach mit. Wird also auch an YouTube liegen. 

Auch Euer Ton gefällt mir ganz gut. Bekommt ihr von jedem Mikro das direkte Signal um die restlichen Mikros stumm zu schalten?
So machen wir es. Sonst hallt es bei uns kräftig.

Nutzt du die Option "Ex comp Mode sel" "NORMAL" oder "Extended"? Extended führt bei uns zu einem recht verrauschten Bild. Habe bisher noch nicht rausgefunden wofür das genau sein soll. 

Mit den VISCA-Commands habe ich schon erfolgreich gearbeitet. Damit machen wir jetzt Zoom-Fahrten. Das ist einfacher als mit dem Keyboard. Mit den VISCA-Commands kann man sich (glaube ich) noch mehr beschäftigen. Das lohnt. Verwendung habe ich auf jeden Fall schon mal verstanden. :-)

dominik34, 2021-04-21:
Ich weiß es nicht genau aber 20m sind es bestimmt....Bei Dunkelheit ist es auch bei uns deutlich schwieriger (siehe Osternacht 😆) allerdings klappt es bei normaler Beleuchtung recht gut. Artefakte hab ich normalerweise nicht. Liegt eventuell an der Kodiereinstellung des PCs?
Ich hab die Erfahrung gemacht dass eine niedrige Bitrate, also starke Komprimierung, vor allem das Bild bei Bewegungen drastisch verschlechtert.

Wenn es von der Beleuchtung her kritisch ist geh ich auf Manuell mach die Blende ganz auf, den Shutter auf 1/30 (minimum bei 30/25 fps) und regle den rest dann entsprechend mit gain (iso).
Unter erweitert gibt es dann noch die Möglichkeit spezielle Einstellungen für extreme Dunkelheit zu verwenden. Bis auf die Osternacht lass ich das aber deaktiviert.
Wenn trotzdem noch etwas Bildrauschen auftritt kann das mit „Noise Reduction“ noch reduziert werden. Extended sel hab ich grundsätzlich aus (also auf normal) keine Ahnung warum das jedesmal von alleine an ist und was es bringen soll. 🤷‍♂️
Was euer Bild betrifft, das ja je nach Sonneneinstrahlung manchmal etwas ausbrennt, könntest du mal die highlightcompensation (hlc) oder unter advanced die Einstellung für „bright“ ausprobieren. Das hat mit geholfen in einer bunt-beleuchteten, dunklen Kirche, die „ausbrennenden“ Flecken der Strahler zu vermindern.

Wegen dem Ton: danke ☺️ 
Selber bin ich aber nicht ganz glücklich damit. Wir greifen den Main Mix der Kirchenanlage ab welcher aber ein sehr starkes NoiseGate drauf hat.....deshalb haben wir ein zweites Großmembranmikro für den Gesang welches wir dann manuell dazuregeln. Dadurch können wir das NoiseGate mit dem von diesem Mikro aufgenommenen Hall etwas überdecken.
Grundsätzlich lassen wir die Mikros der Kirchenanlage aber immer offen allerdings müssen wir die Kirchenanlage beim Gemeinde Gebet zurück regeln da das NoiseGate sonst einzelne Worte heraus schneidet.

In die Visca Commands muss ich mich noch etwas einarbeiten. Hab aber schon die enormen Möglichkeiten dessen in Verbindung mit Companion vor Augen 😊

Unsere Unterhaltung geht nun etwas über das Thema des „Issues“ hinaus. Vielleicht ist es auch für andere interessant. Ansonsten melde dich gerne unter „dospa12(at)gmx(punkt)de“ bei mir.

ruudboon, 2021-05-21:
Not sure if this is solved (don't speak german) but I just received my P200's and will do some testing tomorrow. If needed I can drop a few pulls to solve any issues. Would it be possible to keep the discussions in english to allow more people to help?

wikipeter-de, 2021-05-21:
> Not sure if this is solved (don't speak german) but I just received my P200's and will do some testing tomorrow. If needed I can drop a few pulls to solve any issues. Would it be possible to keep the discussions in english to allow more people to help?
Yes, the problems are solved. If you want to learn and find solutions for problems, there is a very good Facebook Group "BirdDog User Group". There you can get so many help. There is also a good YouTube Video from BD (P200 How To - Setting up and configuration overview https://www.youtube.com/watch?v=LEFsfldWn-Q) and for Pictures settings... https://www.youtube.com/watch?v=frhk2wNjuWk   Check these Videos. And there is also a nice one: https://www.youtube.com/watch?v=frhk2wNjuWk

bryce-seifert, 2022-03-12:
[In beta](https://bitfocus.io/companion/download/builds/) these have been moved to variables, and should display properly on the latest LTS firmware from BirdDog.

closed 2022-03-12 by bryce-seifert