bitfocus/companion-module-birddog-ptz #7: Focus Fine Tuning Request

enca1024, 2021-02-23:
We've have a P4K on loan which we're considering purchasing for a local church and linking up to Bitfocus Companion.  A feature that would be very useful to have is the ability to fine-tune the focus, i.e. move it a certain amount forward or backwards from the current position as the camera doesn't always pick the right focus point.

The Visca command "8x 09 04 48 FF" returns the current focus position in the response "y0 50 0p 0q 0r 0s FF" and you can set a new focus position using "81 01 04 48 0p 0q 0r 0s FF".

Perhaps you could have a "Focus Adjust" command which has "Near" or "Far" options along with amount options "Large", "Medium", "Small" and "Very Small".

When triggered it would read the current focus position and adjust the focus by 0001 (relating to pqrs) for "Very Small", 0010 for "Small", etc. in the right direction.  In my tests, 0000 is far focus and F000 appears to be the max for near.

You'd need to include some basic limiting for situations such as the amount is set to large, direction far and the current focus value is 0123.  0123 minus F000 is <0000 so it just reduces to 0000.  Similarly for situations where the value increases to more than F000.

Note that this function does not work unless Focus Mode is Manual.

Many thanks

nottooloud, 2022-10-05:
Should this still be open? Manual focus works on my P200 with very good control.

jmcc94, 2022-10-18:
Closing dues to no activity

closed 2022-10-18 by jmcc94