bitfocus/companion-module-birddog-ptz #13: Off by one error on preset numbers on P200 cameras.

mtaalas, 2021-07-27:
Equipment: Birddog P200, Birddog PTZ Keyboard, Elgato Stream Deck
Firmware: P200: 21.01.3384, PTZ:  Firmware 1.95, baseboard 0.9
Protocol: Visca-IP

When you tell companion to call "Preset 1" on a P200 camera, it actually calls "Preset 2" that was saved using BD PTZ keyboard etc. This is consistent across all 9 presets tested on latest firmware (21.01.3384) on BD P200 and PTZ Keyboard firmware.
It's propably an "off by one" error, where either this module, or the BD P200 OR PTZ Keyboard starts presets from index 0 and the other starts from index 1 internally and there's a mismatch.

Meaning that when I save preset 1 on the Birddog PTZ keyboard, it's internally an index 0 in the camera firmware etc. But the fact is that BD software or hardware don't seem to know "Preset 0", they all start from 1 from the users perspective.

I haven't looked deeply into this yet (like on low level), maybe after the broadcast, but this might be something that needs to be looked into.

pjbisset, 2022-01-26:
This issue applies to the P100 cameras that I have as well - I compared using NDI Studio Monitor rather than the keyboard with the same result. Module preset # = NDI - 1.

bryce-seifert, 2022-03-08:
Yeah, I think the BirdDog keyboard is the odd one out here. In my testing, Companion, NDI Studio Monitor, and the BirdDog Web UI all return the same thing for Preset "1".

Luckily, this seems to be fixed in the latest firmware version of the keyboard, and now if you try and recall "0" open the keyboard it says out of range. So this should be resolved!

closed 2022-03-08 by bryce-seifert