bitfocus/companion-module-birddog-ptz #19: [FEATURE REQUEST] - Rewrite of BirdDog module to take advantage of new firmware & API

jmcc94, 2022-01-18:
With the release of the LTS firmware for the BirdDog range of devices, it looks like the control mechanisms have changed, as well as a release of a new API.

What would it take to get either this module updated or rewritten to take advantage of the new features (could be combined with companion-module-birddog-studio and just have a single all encompassing one)

- API 2.0 - https://bird-dog.tv/SW/API/index.html
- VISCA over IP document - https://desk.zoho.com/portal/birddogtv/en/kb/articles/command-list-for-p100-viscaip

I also have a API to VISCA values document off BirdDog that I could make available

I have no skill in being able to do this, but I would be willing hire to make this happen

jmcc94, 2022-01-18:
@josephdadams - Is this something you would be interested in?

josephdadams, 2022-01-18:
Sure! Are you in the Bitfocus slack channel?

jmcc94, 2022-01-18:
Yes

jmcc94, 2022-01-18:
Yes I am

josephdadams, 2022-01-18:
Let's chat there about it. Thanks!

tim-barr, 2022-02-03:
+1  Thanks guys

bryce-seifert, 2022-02-07:
I talked to Joseph and I'm going to to a look at getting a module going with the new API.

I also maintain the BirdDog Studio module, but I think it makes sense to keep the camera stuff separate from the encoder/decoder since the actions are different enough and since the encoders haven't gotten their LTS update yet, it seems like they'll be on different release schedules.

Will get started today and hopefully have an update soon!

jmcc94, 2022-02-07:
That's great! 

Did Joseph pass on the thoughts I had on what I wanted done? If not, I'll post them here.

bryce-seifert, 2022-02-07:
If you could post them here that would be great!

tim-barr, 2022-02-07:
While you guys are looking at the LTS for the Cameras and Studio, please see if control of the Keyboard is possible.
Specifically, I would like the ability to "Select" a particular camera from Companion.  Thanks!

bryce-seifert, 2022-02-07:
I don't see the keyboard listed in the API docs, but I'll ask BirdDog if possible

jmcc94, 2022-02-07:
@bryce-seifert - Here you go

---
If I was better at coding, I was thinking of using https://github.com/utopiantools/node-visca as a dependency to take care of the VISCA traffic....

I also got this document from BirdDog support with this message
_"I have attached a document, that was provided by one of my dev team, that lays out all the VISCA output that you are seeing returned from the API. This document was made by Sony, and I think it is for the PF120 imaging unit. I checked it against the values being returned from my P100, and most of them were correct. The main one I noticed being off was the iris readout, but that seemed to be off by a set number of steps, so you should still be able to use these charts once you factor that in.
Thanks for asking this question, as I got to learn a couple of things myself around how the APIs return their values. Including that, the returned values are a Hex number. Enjoy digging through this, and if you need any more information, let me know."_
[API to VISCA values.pdf](https://github.com/bitfocus/companion-module-birddog-visca/files/8012129/API.to.VISCA.values.pdf)

Re changes - As we only have the PF120s & P100s at church, ideally I would like to cover off what those can do, though I know the module would be more useful if it could also work with the P200, P400, A200, A300 & P4K etc...

I think the current VISCA PTZ control seems to be fine, though I would like to add a couple of functions;
- Zoom Direct to a zoom position with custom speed
- Focus Direct to a focus position
- It may also need to be tidied up a bit
  
The other functions that are currently done by VISCA, have those moved to calling the API, and have the responses in variables and available for feedback. I think the functions to initially focus on are;
- Device details
- PTZ Details
- Exposure
- White Balance (though I think 'One Push' trigger still may need to be VISCA)
- Picture Settings

Additional useful ones could be
- NDI Encode details
 - Colour Matrix

...Then the rest of the functions that are used by the rest of the camera range...

Other potential features;
In the config page, would it be possible to have a drop down list of the devices on the network to choose from? The devices announce themselves using bonjour, so hopefully that would work, but it maybe too early in the module load?

I was thinking it might need to be a new module, so as to leave the current visca module to deal with the cameras that are using the old firmware

My coding skills aren't great, but I'm happy to help by doing some typing in of constants etc...

Cheers,
Jamie

tim-barr, 2022-02-07:
Thanks Bryce.

I don't have the new KBD firmware installed, but maybe this new video means more to you than it does to me.
It shows how the Function keys on the Keyboard can be programmed to SEND commands to Vmix, but I also read somewhere that the function keys can be set to select a camera on the keyboard.  So maybe there is an API command it sends to itself to select a camera.
https://www.youtube.com/watch?v=6TlAdQxhy3Y

Note:  My desired functionality is to have the keyboard be auto-selected based on the same streamdeck keys that are putting my cameras into Preview on my Atem.  This way the volunteers won't move the on-air camera and won't have to select anything on the keyboard.

bryce-seifert, 2022-02-08:
@tim-barr FYI on the keyboard component, this was the response from BirdDog
> We haven't released an API package for the Keyboard yet as it is still in development

tim-barr, 2022-02-08:
Thanks for checking!

jmcc94, 2022-02-23:
@bryce-seifert - How are you getting on? Is there anything I can do to help?

bryce-seifert, 2022-02-25:
Still forging ahead, hopefully I'll have a draft by this weekend. It needed a lot more TLC then I thought

bryce-seifert, 2022-03-08:
@jmcc94 Ok I have a little more cleanup to do, but I'm hoping to merge the initial rework later today.

Do you have a dev environment to help test? Also curios what cameras you have, I have a P200 to test but nothing else in the range. Also I might hit you up on slack for easier back and forth once we get testing

jmcc94, 2022-03-09:
@bryce-seifert 

I have access to P100 & PF120, but will need to head into Church to pick them up.

I haven;t gotten around to installing a local dev companion, and then running the updated module in local-dev-module - so will need to wait until it's in beta... if that is going to take a while, I'll get onto installing a dev companion...

bryce-seifert, 2022-03-11:
@jmcc94 Ok the initial rewrite is in the beta builds now. Any testing you have time to do would be appreciated!

jmcc94, 2022-03-13:
@bryce-seifert - I've managed to do a test with the P100, but would like to finish with the PF120 before I give you the full results. Overall it works well - just some VISCA commands not working, and  a few inconsistencies with values compared to actual camera values.

bryce-seifert, 2022-05-02:
Closing this as we are now using the API as of 2.0

closed 2022-05-02 by bryce-seifert