bitfocus/companion-module-birddog-ptz #20: Focus Near/Far don't work in 2.1.4 release of Companion

pjbisset, 2022-01-26:
In Companion 2.1.3 Focus Near/Far worked, but the spelling mistake on Focus (Forcus) Stop meant I could not used them they way I wanted with the release calling Stop. In 2.1.4 of Companion both Focus Near/Far no longer do anything. I have tried them with no other action and they make no change to the camera's focus state.

I am using the P100 camera with firmware v21.01.3766.

bryce-seifert, 2022-03-08:
Can you verify that your focus mode is set to Manual? Otherwise, the Focus Near/Far won't work in auto.

pjbisset, 2022-03-09:
Okay. Give a few days to organise a visit or remote access.

pjbisset, 2022-03-21:
Sorry for the lengthy delay. Confirmed autofocus NOT on - focus far/near still no good.

bryce-seifert, 2022-03-24:
Bummer. Are you using the latest LTS firmware for the camera? If so, can you try Companion 2.2.0 and see if it works there?

pjbisset, 2022-03-25:
Hi, bit cautious about updating the firmware and losing compatibility with Companion - I'm one version behind on the firmware.
Will think about options and respond. Might take a little while to organise and test.
Peter

pjbisset, 2022-04-22:
Finally able to test with (almost) latest f/w v4.5.3787-LTS and beta build of companion companion-win64-2.2.0+4105-beta-662c35f5. Focus Stop command named correctly (I didn't check 2.2.0 release before trying the beta to solve the frame rate refresh issue) and focus change stops on release of button as expected. Very fine grained control now seems to be working perfectly. Nice work. Thanks

closed 2022-04-22 by bryce-seifert