bitfocus/companion-module-birddog-ptz #31: BirdDog framerate resetting every 3 minutes

AndyVegetable, 2022-03-26:
Minor but in 2.2.0. Still in 2.2.0+4038-beta-835ba58b.
Since upgrading, when the "BirdDog PTZ" module is enabled, our camera resets to 1080p60 instead of our normal setting of 1080p29.97.

After resetting the framerate, it takes about three minutes for the frame rate to change back to 60.

If I disable the "BirdDog PTZ" module, the frame rate remains at 29.97. I did try removing and re-adding the module.

Our BirdDog P200 is on firmware 4.5.3418-LTS.

When the frame rate changes, the BirdDog video no longer shows up in our ATEM, since its frame rate is set to 29.97. Temp solutions, either stop using the BirdDog for streaming, or not use Companion for controlling it.

swperic, 2022-03-26:
I second this.

bryce-seifert, 2022-03-28:
After looking into this, I think I've narrowed this to a bug with BirdDog's API 2.0. When I completely turn off API polling, I no longer get this issue.

I've submitted a ticket to them to see what might be happening.

AndyVegetable, 2022-03-28:
Excellent,thank you!

AndyVegetable, 2022-04-04:
Is BirdDog playing nice? Is the issue a tough nut to crack?

bryce-seifert, 2022-04-05:
I think I've narrowed it down to one API command, and I'm still in contact with them.
Going to try and work around this command for now.

AndyVegetable, 2022-04-15:
Sorry that this is so difficult. Thank you for working on it!

AndyVegetable, 2022-04-20:
Any luck?

bryce-seifert, 2022-04-20:
BirdDog has indicated it’s on there list but no timeline for a fix. And I’m still waiting for a reply on using VISCA to get frame rate from them.

I think I’ll work on just avoiding this call for now and try and get that fix out this week

AndyVegetable, 2022-04-21:
Thank you for your time and effort!

bryce-seifert, 2022-04-22:
@AndyVegetable If you have some time, please try the latest Companion beta. While BirdDog hasn't solved the API issue, we've temporary removed it from polling so this should prevent the frame rate switching issue. Let us know if it still happens or you encounter anything else.

I recommend backing up your Companion config first, and then [installing the beta from here](https://bitfocus.io/companion/download/builds/).

pjbisset, 2022-04-22:
This appears to have solved the issue for me. I have P100s on f/w 4.5.3787-LTS and had the original issue. Beta build does not cause the frame rate change.

AndyVegetable, 2022-04-23:
I will try it this weekend. Thank you!

AndyVegetable, 2022-04-25:
> @AndyVegetable If you have some time, please try the latest Companion beta. While BirdDog hasn't solved the API issue, we've temporary removed it from polling so this should prevent the frame rate switching issue. Let us know if it still happens or you encounter anything else.

It works. Thank you!

bryce-seifert, 2022-04-28:
Glad to hear it. I'll close this for now, and if BirdDog fixes their API we'll implement a more permanent fix

closed 2022-04-28 by bryce-seifert

Ghostbird, 2024-06-30:
Sorry to resume such an old issue, but we've got this problem on our BirdDog P200.

- Latest Companion 3.4.0 (3.4.0+7064-main-b5ad56df)
- Problem seems to be gone when we disable the BirdDog connection in Companion
- Doesn't happen every three minutes, but happens unpredictably, but fairly regularly.

This means that currently we cannot control the BirdDog P200 through Companion, because that could cause it to drop off-stream at any moment. It would be nice to have the option to have a set of named presets as buttons in Companion.

bryce-seifert, 2024-06-30:
I would recommend updating to the latest Si2 / firmware v5. That should resolve this specific issue. There are some v5 specific issues that we're also waiting to be fixed in the API (see https://github.com/bitfocus/companion-module-birddog-ptz/issues/131), but the frame rate issue should be resolved.

Despite being "LTS" it doesn't seem like FW v4 is getting any more support, give the last release was almost 2 years ago.

Ghostbird, 2024-06-30:
Ah, thanks. We had assumed that since the P200 Si2 was a different physical camera, that firmware would not apply to the P200, and that there simply was no newer firmware for our camera. I just dove a bit more into the 5+ firmware release files and saw that since  version 5 they've made a unified firmware that applies to all cameras.

I found out that the 4.3 LTS is not there as a service release that still receives updates, it's there as a transition firmware. Older versions can safely update to 4.3 LTS and from 4.3LTS you can safely upgrade to 5+. Direct upgrades from older versions to 5+ will not work.

bryce-seifert, 2024-07-01:
No worries, I agree, it is a bit confusing. Let me know if you need any additional help!

Ghostbird, 2024-07-10:
Since the update the issue got much worse. You can change the video mode, and the BirdDog will report that it's running at 1080p50, in both the UI, and in Companion. However the SDI will actually still output 1080p60, until you disable the BirdDog connection in Companion.

Do you have any idea which exact API call to the BirdDog might cause the issue? I contacted BirdDog, but their engineers consider this a Companion problem, since that's the only software that faces the issue.

If I can construct a small script that provides a minimal reproduction scenario, I can convince them to look into it.

bryce-seifert, 2024-07-10:
Can you confirm the firmware version you are on?

When we tackled this 2 years ago with their engineers the issue was the encodesetup endpoint, and we did what you were suggesting: made a separate script that demos this behavior outside of companion. I can try digging up that file soon

Ghostbird, 2024-07-10:
I've checked the versions:

Firmware: BirdDog P200A4_A5 5.5.114
MCU: 33

Ghostbird, 2024-07-10:
Note: We tried several options to get it to work:

- Set the Video Mode
- Set the Video Mode and _Restart_ the device
- Set the Video Mode and _Reboot_ the device
- Set the Video Mode and physically power cycle the device

In all cases the video mode was shown in the UI as 1080p50, and also reported in companion as 1080p50, but the SDI output was only visible in our BlackMagic Smart VideoHub 40×40, which can decode and route SDI regardless of video mode (within reason). And it showed that the SDI video mode was 1080p60. Other devices in the network could not decode the video since our studio is configured for 1080p50 production.

Once we disabled the BirdDog connection in Companion, the video shortly after automatically showed up on the other devices, and the video hub showed that the SDI now was 1080p50.

bryce-seifert, 2024-07-14:
I found the test page I built to test this outside of Companion initially. You can download it [here](https://www.dropbox.com/scl/fi/mtje9t65iflquu752xtl6/birddogtestpage.html?rlkey=y6dcxnua278hhf9mh8kh08ct2&st=wlqgd7ik&dl=0).

Possibly you can see if the issue persists while running this? I'm not sure if it's still encodesetup causing issues, or what it might be. 

But, to my knowledge, we don't have the power with the API to control SDI output format separately than the overall output format, so if the issue is caused by Companion being on, it's likely something on the BirdDog API end because we couldn't intentionally or accidentally be adjusting the SDI format separately.

Ghostbird, 2024-07-15:
Thanks! I'll give this a try.