bitfocus/companion-module-birddog-ptz #56: Latest build preset and comm issues

obcocav, 2022-05-29:
Good morning,

It seems in the latest build that the preset recall command is maybe only reading the first digit in a two digit preset number.  For example if I recall Preset 12 the camera actually recalls preset 1.  Same if i set that preset.  Preset 12 is stored as #1.  
I have also had a few instances where the camera quits responding for a time.

Any ideas?
Thanks!

jmcc94, 2022-05-30:
What FW are you running on the camera?

The module has been rewritten to use the API of the cameras, and it seems as though the API is limited to only accepts 9 presets.

I'm trying to confirm this with BirdDog themselves

jmcc94, 2022-05-30:
@obcocav - The latest birddog-ptz module v2.1.1 was added in beta 4189, so if it was all working in 4197 as you say, then I wonder if it's something to do with some of the subsequent updates to companion.

The testing I've done seems to suggest that the API is limited to only 9 presets, but am talking to BirdDog about that now. Maybe in the forthcoming new Firmware...

obcocav, 2022-05-30:
Yes sorry.  Wast thinking current up to that point.  I also recently upgraded to the 4318 LTS firmware so that would make sense too.   Why in the world would they go backwards like that!  Seems that would have to be a glitch not an intentional decision?  Even their keyboard supports more than 9 presets.

obcocav, 2022-06-03:
@jmcc94 .  Thank you!  Have been talking to their support and was told this was not an issue and to set my cams gateway to my PC…..that obviously is not correct and bad advice.  Hope your contact is right and the update comes soon!  This is showstopper for us.  We lose connection at the worst times…

Thanks for your work!

obcocav, 2022-06-03:
@jmcc94 .  For reference.  I went back to Beta 4127 and all functions work as expected!  Thats just a previous version I had handy but that is with 3418 Still on the camera.  Does that mean anything to you in tracking down the issue?

Thanks!
Kevin

obcocav, 2022-06-05:
@jmcc94 .  Another data point.   I think it may be companion locking up the camera.  I'm back on 4127 still and having no issues at all.  Even my other control software (Rocosoft PTZjoy) is working fine and the camera is never not responding.  Presets work correctly.  When I upgrade to the newer betas all the comm issues return along with the preset issues.  Other control protocols separate from Companion quit functioning after a time as well.

obcocav, 2022-06-12:
@jmcc94.  Just wondering if there was any update with this and making sure you saw my previous post.  4127 is working great with LTS 3418 (Latest firmware) so it appears to not all be the Camera firmware.   Again, thanks for all the work yall do with this module and Companion!  Hope this info is helpful!

jmcc94, 2022-06-13:
@obcocav - can you confirm what version of the birddog-ptz module is in beta 4127.... I can't remember what ver went into what beta , and that beta isn't # available for download anymore...

obcocav, 2022-06-13:
@jmcc94.  

It 2.0.1 in that version.

And yes I can save and recall any number of presets correctly.  Also set just tried deleting and re-adding the module on the latest beta with no change.  Setting and recalling 11 actually sets and recalls 1 etc.  Thanks!

obcocav, 2022-06-19:
@jmcc94 .  Hopefully the next firmware is very soon!  Guess we'll stick to the older version to keep things running but very excited for a couple of new companion features in the newer builds.  Thank you!

ervings, 2022-06-29:
Hi!
There is now a new firmware for p200 and other birddog cameras - after the upgrade, companion does not offer any presets, but the device shows up as connected. 
Anybody has the same issue?

obcocav, 2022-06-30:
@jmcc94 .  Is this latest firmware the one that should communicate with the API correctly?  Thanks.

obcocav, 2022-07-02:
Happy to report this is indeed solved with build 4510 and the NDI5 firmware from BirdDog!

closed 2022-07-02 by obcocav