bitfocus/companion-module-birddog-ptz #58: No camera- actions available

Tilltv, 2022-06-10:
Hi,
the Companion- module is activated and the camera is connected. However, absolutely no actions can be selected.
The camera does not exist in the dropdown menu for creating new actions.
Reinstalling Companion didn't fix the problem.

my second system with the older Companion Version 2.2.0 works fine.

does anyone have an idea?
Thanks!

Windows11
Companion Version: 2.3.0 (2.3.0+4470-beta-00bb68c8)
Birddog Camera: P4K

bryce-seifert, 2022-06-10:
@Tilltv Are there any reported errors in the log page of Companion?

Tilltv, 2022-06-10:
@bryce-seifert No None. only the connection message:
22.06.10 16:42:24 instance(BirdDog PTZ): Connected to p4k

jmcc94, 2022-06-10:
@Tilltv - if you go to this webpage of the camera <camera_ip>:8080/about.... what is the FirmwareVersion value?

The module tailors the actions based on the model, and I wonder if it's not detecting your model correctly...

Tilltv, 2022-06-11:
@jmcc94 
The firmware version is 4.5.212-LTS
MCU- Firmware:V0K190B35010AA010T2

jmcc94, 2022-06-13:
@Tilltv - thanks for that, though I think you missed what I was asking (obviously didn't do a good enough job).

The module talks to the API of the camera to get the information, and the webpage I wanted you to go to is one of the endpoints, and returns a number of fields - one of which is the FirmwareVersion which we currently use to get the Model from.

If you can go to the webpage `http://<camera_ip>:8080/about` and let me know what is returned.

Thanks

BenJamesAndo, 2022-06-13:
I'm also having the same problem. Companion 2.2.0 works but 2.2.1 & 2.2.2 doesn't display any actions or presets.
`{"FirmwareVersion":"BirdDog P400 4.5.212-LTS","Format":"CAM 1","HostName":"birddog-cam3","IPAddress":"192.168.1.144","NetworkConfigMethod":"static","NetworkMask":"255.255.255.0","SerialNumber":"XXXXXXXXXXX","Status":"active"`

jmcc94, 2022-06-14:
@Tilltv  & @BenJamesAndo - would it be possible to get a copy of each of you companion logs - under the log tab in companion, there should be an export log button.

BenJamesAndo, 2022-06-14:
Working on 2.2.0 stable with Birddog PTZ version v2.0.2. Not working on version 2.2.2 stable with Birddog PTZ v2.1.1

Attached is the log from version 2.2.2
[Not Working-GCTV-Broadcast-PC_companion_log_20220614-1536.csv](https://github.com/bitfocus/companion-module-birddog-ptz/files/8896694/Not.Working-GCTV-Broadcast-PC_companion_log_20220614-1536.csv)

jmcc94, 2022-06-15:
Unfortunately, I'm travelling for work, and don't have access to my test environment. The log you provided looks as though it doesn't detect the camera model.... I'll look at pushing out a new module with some additional logging to see if we can set what's going on..

BenJamesAndo, 2022-06-15:
If it helps, here's my log from version 2.2.0 stable where Birddog PTZ presets are working.
[Working-GCTV-Broadcast-PC_companion_log_20220614-1531.csv](https://github.com/bitfocus/companion-module-birddog-ptz/files/8906195/Working-GCTV-Broadcast-PC_companion_log_20220614-1531.csv)

Tilltv, 2022-06-20:
here is the missing information:

**Running system:**
Companion- Version: 2.2.0 (2.2.0-c649d4bd-3529)
BirdDog VISCA Modul- Version: 1.0.4 - I think (version number from the log. There is no Information in the modul- Information chart)
No Log Errors (missing export function)

**Not running system:**
Companion- Version: 2.3.0 (2.3.0+4470-beta-00bb68c8)
BirdDog PTZ Modul- Version: 2.1.1
Log: [Tills_companion_log_20220620-1256.csv](https://github.com/bitfocus/companion-module-birddog-ptz/files/8940075/Tills_companion_log_20220620-1256.csv)

**Camera:**
Birddog P4K with the newese Firmware Version (4.5.212-LTS)

Connected Camera:
[image]

No Camera- actions available
[image]

I hope this helps!

bryce-seifert, 2022-06-21:
I don't have a P400 or 4K, but I was doing some testing feeding it that model and I think I may have found the issue during building actions.

Going to have @jmcc94 double check my work before I merge and then it should be able to test. Thanks for your patience

Tilltv, 2022-06-27:
Hey @bryce-seifert , is it possible to generate a beta version (2.1.2). I would like to try if everything works now.
Thanks very much!

bryce-seifert, 2022-06-27:
@Tilltv Absolutely! I have a pending request to get this in the beta, and I’ll let you know when it’s merged by the core dev team

BenJamesAndo, 2022-07-01:
@bryce-seifert I just tried the new beta build. Unfortunately the problem is still there for me. Still no presets. I did notice Birddog just released a new beta build for the cameras. Perhaps next week I'll try that and see if that helps in anyway.

Attached is a new log file.

[GCTV-Broadcast-PC_companion_log_20220701-1835.csv](https://github.com/bitfocus/companion-module-birddog-ptz/files/9027426/GCTV-Broadcast-PC_companion_log_20220701-1835.csv)

bryce-seifert, 2022-07-01:
@BenJamesAndo Sorry about that. Just to clarify, are actions working now? Is it just presets that don't load? And do you see the BirdDog listed here or it's just blank?
[image]
?

BenJamesAndo, 2022-07-02:
@bryce-seifert no actions are showing up or working, neither are the presets. I can't see the BirdDog module listed, it's just blank.

bryce-seifert, 2022-07-02:
@BenJamesAndo That is super odd behavior. Are you able to create a new BirdDog module, or is that also blank? Can you also do a Log > Export Support Bundle as well and upload that?

Tilltv, 2022-07-03:
Now I finally had the opportunity to test:
At first everything worked. The camera could connect, actions were available and the camera could be controlled.
However, this only worked with the camera- connection that I had created in the previous version. To test, I deleted the camera connection and tried to create it again (new version). Now the error is back. No camera actions are displayed.

Very strange.

[Tills_companion_log_20220703-1102.csv](https://github.com/bitfocus/companion-module-birddog-ptz/files/9034420/Tills_companion_log_20220703-1102.csv)

Tilltv, 2022-07-03:
@bryce-seifert 
New insight: With a lot of connection activation/deactivation it worked again. I created a log with two cameras. One (PTZ2) works, the second (BirdDog PTZ) doesn't work. The hardware is the same. I only have one module activated at a time. Maybe that will help.
[Tills_companion_log_20220703-1440.csv](https://github.com/bitfocus/companion-module-birddog-ptz/files/9034789/Tills_companion_log_20220703-1440.csv)

bryce-seifert, 2022-07-03:
@Tilltv Interesting, it seems like the API is not returning the model every time, which is throwing the `Unrecognized camera model: undefined` Sorry to trouble you, would you also be able to do a Log > Export Support Bundle as well and upload that to get a bit more info.

Tilltv, 2022-07-04:
@bryce-seifert I sent you the data via email.

BenJamesAndo, 2022-07-10:
@Tilltv Yes I have the same thing. Restarting the camera does lead to it showing up briefly sometimes. I didn't get time, but I will definitely update the firmware on Tuesday, and I'll let you know if there are any improvements.

BenJamesAndo, 2022-07-11:
@Tilltv I updated to the beta firmware. Upon starting Companion the logs say `instance(BirdDog PTZ) error Unrecognized camera model: EYESP400`
I then decided to rollback to the firmware before the current LTS version. It was released around June last year. Upon starting Companion the logs said I needed to update to the most recent LTS version for it to work. So now I'm back on the LTS firmware and the camera actions are still not available.

For the meantime I'm now using the generic Sony VISCA module. I put in camera id as id 1, entered in the IP address of the P400 and it worked.
@bryce-seifert attached is the log from when I was running the beta firmware. I also exported the support bundle if that helps.
[GCTV-Broadcast-PC_companion_log_20220711-1536.csv](https://github.com/bitfocus/companion-module-birddog-ptz/files/9087844/GCTV-Broadcast-PC_companion_log_20220711-1536.csv)

jmcc94, 2022-07-11:
@BenJamesAndo - As part of the module providing the appropriate actions for each module, the module queries the camera for it's model. 
Currently it looks like the camera is returning a model that we didn't expect (EYESP400) - we are expecting just P400. 

Just to confirm, are you able to provide the response to the call to the API http://<camera ip>:8080/about

We have a fix in that will allow you to specify the model as part of the config- just waiting for the core devs to put it into companion, and provide a new beta

jmcc94, 2022-07-11:
@BenJamesAndo - can you please confirm a couple of things;
- Camera FW that you have tried
- Companion versio
- Biddog-PTZ module version

Thanks

BenJamesAndo, 2022-07-12:
@jmcc94  The API response I get is: `{"FirmwareVersion":"BirdDog P400 4.5.212-LTS","Format":"CAM 1","HostName":"birddog-cam3","IPAddress":"192.168.1.144","NetworkConfigMethod":"static","NetworkMask":"255.255.255.0","SerialNumber":"70b3d53bxxxx","Status":"active"}`

Yeah I guess the new beta firmware now calls the P400, EYESP400. 

Being able to manually pick the model in companion sounds great. I'll definitely test it once it's ready.

I've mostly tried on Firmware BD_P400_4.5.212-LTS. This is my current firmware that I'm running.
I tested the beta BD-P400-5.0.054-Beta
As well as the firmware before 4.5 LTS which was P400 21.06.195

Running on companion-win64-2.3.0+4532-beta-6890cb37
Birddog-PTZ module 2.1.2

jmcc94, 2022-07-15:
@Tilltv / @BenJamesAndo 

The latest companion beta now has the latest birddog-ptz module, which allows you to set the model in the config.

BenJamesAndo, 2022-07-17:
@jmcc94 I did some quick testing today and it seems to work really well. I set it manually to P400 and all the actions and presets showed up and I was able to control the camera. Thanks for this fix.

Tilltv, 2022-07-18:
Also works great with the P4K!
Great work! Thank you for the fix!