bitfocus/companion-module-birddog-ptz #62: Module caused cameras to enter “decode” setting

rjurden, 2022-07-10:
I have four P200 cameras. I had the companion instances disabled to troubleshoot a different issue. When reenabling the instances, three of the cameras immediately lost their picture over NDI and began displaying in the web UI that they were in “decode” setting instead of “encode” setting. 

This did not affect the fourth camera.

The cameras were completely unresponsive. Power cycling did not resolve the issue, the OSD menu would not come on using the remote and a monitor. Could not use our external PTZ controller.

The solution was to re-flash the NDI firmware from the web UI.

I believe the companion module caused this because it happened instantaneously upon reenabling the instance for each camera.

Attached are screenshots of a camera that was working (encode settings displayed) and a camera that was locked up (decode settings displayed). 

Running the latest Birddog firmware.
 
[image]
[image]

rjurden, 2022-07-10:
Another user in the Birddog Facebook group reports that he had the same issue with one of his P200 cameras and that he is also using Companion.

bryce-seifert, 2022-07-10:
Looking into this. We've been testing with the BirdDog beta firmware and haven't seen this, so far, but will try to replicate. Just to confirm, what version of Companion are you on? Are there any errors in the companion logs?

rjurden, 2022-07-10:
I used 2.3.0+4506-beta-86c5c823
<https://bitfocus.io/companion/download/build/companion-win64-2.3.0+4506-beta-86c5c823.exe>

I will check the logs again when I am back at the computer and have the
issue happen again. This may not be for another week

Ryan

schwtyl, 2022-07-12:
> Another user in the Birddog Facebook group reports that he had the same issue with one of his P200 cameras and that he is also using Companion.

It me. I'm pretty sure we are using the latest stable Companion release, though it could still be a Beta release from troubleshooting https://github.com/bitfocus/companion-module-birddog-ptz/issues/60

I ended up reverting the P200 back to v4 stable FW (I'd upgraded so that line-level analog audio input would work, but it was still sub-par audio so we moved audio to Dante, separately)

atscomms, 2022-07-19:
I have two P200's on a job right now. One as the new 5.0.054beta fw, the other has the release version. After connecting them to Companion...
The beta cam exhibited the same symptom, but it said Encode mode, Changing Video Format, and had no NDI video.
The other camera did not have that problem, but try any PTZ move and it would automatically return to either the Home position or preset 1.
Drove me crazy for 8.5 hours!
Using Companion Ver.   2.2.0 (2.2.0+4071-beta-5ee4e5e7)

jmcc94, 2022-07-19:
@atscomms Please try a newer build of Companion, we implemented fixes for a number of bugs that seem to be associated with the BirdDog API. 

These will be Companion Beta builds - https://bitfocus.io/companion/download/builds/

atscomms, 2022-07-19:
HiI tried the latest zoomiso beta build from today. It didn't kill the ndi video on the first camera,  but it did affect the ptz of my second camera.Any manual ptz movement is promptly returned to a Home position. ArtySent from my Veri Sammy Galaxy

schwtyl, 2022-07-21:
@atscomms Can you confirm the firmware of the second P200? It needs to be v4 (specifically look for LTS in the name). My P200's were exhibiting a behavior similar to what you mention (reverts to preset 1 unexpectedly) when I tried using Companion with a pre-LTS firmware P200.

> but it did affect the ptz of my second camera.Any manual ptz movement is promptly returned to a Home position.

atscomms, 2022-07-22:
HiThey are packed up right now,  but they are this year's vintage,  and I've confirmed that the are A4/5 cameras.Sent from my Veri Sammy Galaxy

obcocav, 2022-08-05:
I thought this issue was fixed using the latest beta Companion and beta 5.0NDI firmware for the p200 but came in yesterday and it returned.  Camera was "corrupted" with no video feed and the stream name changed to "TREAM" again.  Refreshing the step 2 of the NDI firmware on the camera fixes it every time.  Not sure which end this is on as both are beta but if I can help trace it down please let me know!  This is a showstopper for us.  Thanks for all yalls work!

bryce-seifert, 2022-08-06:
@obcocav We have been working with BirdDog on a fix for this, as it’s on the firmware side. Not sure when it will be released as it will have to be a new firmware version. I’m going to DM you on slack

jmcc94, 2022-08-28:
Version 2.2.0 of the module has just been put into the latest beta build of companion.

We have implement some changes to improve the communications with the camera, as well as working with BirdDog to implement fixes to these issues.

We have also added support for v5 FW and the new cameras.

I'll close this issue, as I belive this should be fixed in this release, but obviously let us know if it hasn't

closed 2022-08-28 by jmcc94

nottooloud, 2022-10-04:
I just hit the TREAM thing today. P200 is nonresponsive. I'm on firmware BD-P200-5.0.054-Beta and companion-mac-arm64-2.4.0+4710-beta-05bf0507 running on an M1 Mini.