bitfocus/companion-module-birddog-ptz #67: Connection but no press action

BW-TH, 2022-07-28:
i have a connection to birddog, but when i use it for an press action the bird dog connection is not liste
[image]
[image]
d

jmcc94, 2022-07-28:
@BW-TH. There's been an issue with detecting the model of the camera with the newer models or some models running the latest FW.

This has been partly fixed in the latest beta of the model, which will be in the latest beta of Companion. 

What model is it, running what firmware, and what version of Companion?

BW-TH, 2022-07-28:
Companion V 2.2.3 running on Raspberry and on Windows 11
Birddog P4K: 4.5.212-LTS

know i have imported an old version of configuration, and there the birddog button work, but is not editable

jmcc94, 2022-08-28:
@BW-TH There is a new version of the module in the latest beta of companion that has improved detection of both camera & FW.

closed 2022-08-28 by jmcc94