bitfocus/companion-module-birddog-ptz #68: BirdDog Module not recognizing model P120

AndyVegetable, 2022-08-18:
Just upgraded to Companion 2.3. One of our BirdDog instances wasn't connecting. I checked the log and saw "instance(BirdDog PTZ): Could not connect, unrecognized camera model: P120".

It had connected in 2.2 (before 2.2.3, in that version I noticed that it said it connected, but actually wouldn't do anything. I saw this version, so tried that instead of opening a ticket on old code.

bryce-seifert, 2022-08-18:
@AndyVegetable We have been waiting on documentation from BirdDog about the API parameters for those new camera models so we can add them. We actually just got it yesterday, so we'll be working on full support soon. 

The reason it "connected" in 2.2 but didn't work is because that older version didn't have any logic to detect if a model was unsupported.

jmcc94, 2022-08-25:
Closed by #63

closed 2022-08-25 by jmcc94