bitfocus/companion-module-birddog-ptz #69: P200 tally lights on/off

asukumite, 2022-08-23:
In previous versions of the module we were able to turn the tally lights for P200 on and off.

Currently there is only the option to turn Tally Mode on/off.

jmcc94, 2022-08-24:
@asukumite - Do you know what module version it would have been?

The version 2 and later of the module have been moved away from using VISCA to control the camera, towards using the API.

I suspect it was a VISCA command, as I'm not sure there is a specific command in the API to turn the Tally On/Off.

Will check though.

asukumite, 2022-08-24:
Thanks!

I’m not sure of the module - the change did happen sometime in the past 4 months though.

jmcc94, 2022-08-24:
Can I also confirm the FW of the camera?

Has it been updated since you last had the ability for Tally on/off...

asukumite, 2022-08-24:
Yes, I’ve updated to the 5.0.034 beta.

jmcc94, 2022-08-24:
It was definitely a VISCA command that doesn't seem to be replicated in the API.

I'm also thinking things might have changed between the FW releases...

Anyway, you should be able to test... if you create a Custom VISCA action and put the following codes in the field;

Tally Light On - 81017E010A0002FF
Tally Light Off - 81017E010A0003FF

I found with brief testing with my PF120 that I couldn't turn it on , but could turn it off - I needed to go into the OSD of the camera and turn Tally On from in there..... but that might be a PF120 thing...

If that works for you, you can either leave them as custom actions, or we can look at permanently adding those actions

jmcc94, 2022-08-28:
@asukumite 

Further testing shows that once the Tally mode is on, then 

- Tally Light Green - 81017E010A0001FF
- Tally Light Red - 81017E010A0002FF
- Tally Light Off - 81017E010A0003FF

asukumite, 2022-08-30:
That worked - thanks so much!

jmcc94, 2022-08-30:
Good to hear... will close this issue

closed 2022-08-30 by jmcc94