bitfocus/companion-module-birddog-ptz #71: Camera Actions Not Available in Button Action Dropdown

Dev1082, 2022-08-28:
In the latest beta build of Companion, 2.3.0+4635-beta-66b32bce, I am able to add a BirdDog PTZ connection, connect to a camera, and see an "OK" status in the connections tab. However, when I go to select a BirdDog PTZ action as a button press action, no actions are available. This appears to be a BirdDog PTZ module specific issue as I am unable to reproduce the issue with other modules in this beta build.

[image]
[image]
[image]

jmcc94, 2022-08-28:
It looks like the module is not detecting  the camera model properly...

What camera and FW are you using?

What does the log tab say?

You can manually specify the model in the config, rather than auto detecting the model... Does the problem persist if you do that?

Dev1082, 2022-08-28:
I'm using a P120 running firmware version 5.0.054; this does appear to be the latest version available.

The Companion log does not show anything related to the BirdDog PTZ module. 

The problem does persist when manually specifying the camera model in the config.

jmcc94, 2022-08-28:
Ok - are you able to export the log from the 'Log Tab', and upload it here...

Dev1082, 2022-08-28:
Log file is attached. In this log capture timeframe, I deleted my PTZ connection and added it again. First added as an automatic detection of the camera model, then I edited the connection to specify the camera model as a P120 and renamed the connection to "CAM1."

[companion_log_20220828-1711.csv](https://github.com/bitfocus/companion-module-birddog-ptz/files/9440569/companion_log_20220828-1711.csv)

jmcc94, 2022-08-28:
What happens if you specify you camera as a P100... do the actions now come up?

Dev1082, 2022-08-28:
They do come up when selecting P100. I had tried a few camera models unsuccessfully but not the P100. In limited testing so far, it appears to work fine using P100 as the model for now. Thanks for the assistance.

jmcc94, 2022-08-28:
Yeah the P110 and P120 are just better P100's, so basically all the actions and variables should be the same (minus a few small ones)...

the module must be failing in generating the actions for the P110 (and probably the P120).. .will take a look, and see if I can get a fix out...

Dev1082, 2022-08-28:
Thanks for the help in getting it working. Just to be clear, I'm using a P120 so I can confirm the actions don't generate for that model, and probably the P110 although I don't have one to test with.

jmcc94, 2022-08-29:
Sorry - got them the wrong way around

jmcc94, 2022-08-29:
PR #72 fixes.

Have requested merging into companion beta

Dev1082, 2022-08-29:
I will give the fix a try as soon as it gets merged into Companion beta.

For what it is worth, I've run into an issue where connecting to the P120 using the P100 model setting triggers the decode behavior observed in Issue #62. The connection itself doesn't trigger the issue, but it seems to pop up as soon as a command is sent. Will see if this behavior is replicated with the P120 fix or not.

jmcc94, 2022-08-29:
That's a FW issue and is fixed from 0.055 onwards... You should be able to ask BirdDog support for access to 0.057 (which I think is the latest)

jmcc94, 2022-09-04:
@Dev1082 the latest companion beta now contains the fix for P110 & P120

closed 2022-09-04 by jmcc94