bitfocus/companion-module-birddog-ptz #78: PTZ speed controls

nottooloud, 2022-10-05:
I made a set of buttons with 3 actions each, setting P, T, and Z speeds all to small, medium, or large values. It never works to push a button just once, need to push it 3 times at least, and often more, before the camera will take all 3 values. I tried 5 ms delays between the actions.

P200 firmware 5.0.059 and companion-mac-arm64-2.4.0+4710-beta-05bf0507.

jmcc94, 2022-10-18:
@nottooloud will take a look, but I don't have any P200 to test with, only P100. 

Does the problem still exist on the recent release 5.0.064?

joakyhsf69, 2022-10-20:
Yes, I have several P200 with firmware 5.0.0.64 and I have the same problem

obcocav, 2022-10-22:
Can confirm same issue with P200 A2 and A4 revisions on 5.0.0.64 and latest beta Companion.

jmcc94, 2022-10-25:
When you say '...before the camera will take all 3 values...' how do you know?

Are you using variables or checking the BirdUI on the camera?

jmcc94, 2022-10-25:
@nottooloud - As BirdUI / CamControl and Companion all poll the API to get the information, there can be a delay between setting a value, and having that value represented in the visualization.. and, as you say, the BirdUI doesn't auto-refresh.

From what I'm seeing with my testing, the speed changes are behaving similarly to what I would expect ... I'm seeing almost instant change in the companion variables... which polls the camera API

BBrazner, 2022-11-01:
I'm having success with minimum 100ms delay between actions involving p/t/z speed.  Also, once the camera accepts a speed command, the feedback variables update right away.

bryce-seifert, 2023-03-19:
This has been narrowed down to an issue with the BirdDog firmware itself, as it is reproducible outside of Companion. 
We have informed BirdDog of the issue.

The temporary solution is to add a relative delay between each action.