bitfocus/companion-module-birddog-ptz #79: Suspected cause of loss of camera PTZ control

BBrazner, 2022-10-12:
I have three P200 cameras all at 4.5.3418-LTS. All three send NDI to vMix (v25.0.0.29). All three have PTZ presets in vMix (VISCA IP). This works fine. I also have this Companion module for all three cameras [2.3.0 (2.3.0+4608-v2-3.0-2994a6d0)]. This too works fine, but ...

**Every one or two days**, a camera will no longer respond to PTZ control commands. The problem has occurred on all three cameras at times. When the problem occurs on a camera, the camera is no longer controllable by vMix VISCA IP, nor by the BirdDog PTZ keyboard (also VISCA IP), nor by Studio Monitor (NDI). Upon rebooting the camera (via the BD WebUI), the camera becomes controllable again by all three sources just mentioned. 

I believe the problem started when I upgraded to 2.3.0.  I disabled the module for all three cameras **seven days** ago and the problem has not recurred since.

I don't have conclusive evidence that this module is the cause of the problem, but since the problem was occurring every 1 or 2 days when the modules were enabled, but has not recurred in the last 7 days with the modules disabled, I have strong belief that the module is at fault. 

My next step is enable the module for just one of the three cameras to see if the problem returns for just that camera.

jmcc94, 2022-10-18:
Hi, are you able to confirm what version of the birddog-ptz module is running.

We've done a few updates, and I can't remember which module is in which version of companion.

Are you stuck on using the LTS firmware, or would you consider moving to v5?

jmcc94, 2022-10-18:
@BBrazner ... it looks like that version of companion is running v2.1.3 of the birddog-ptz module.

There was a significant rewrite for v2.2.0 which is in companion beta from v2.3.0+4634beta...

I think it maybe in the v2.3.1 release version, but not sure.

I would recommend upgrading your version of companion to see if at least that version of the module solves your problem.

BBrazner, 2022-10-19:
I will update to v2.3.1 Companion and will get back to you.

Side note: I've been on LTS as the BD NDI 5 has been in beta and I was waiting for a final release. It now appears (as of yesterday Oct. 17?) that it has finally happened, so I'll be updating the cameras, too.  Stay tuned.

BBrazner, 2022-10-20:
I updated Companion to 2.3.1 and enabled the module for just one of my 3 P200 cameras. I also noticed that my pan, tilt and zoom buttons issued vMix commands (e.g., "PTZMoveDown Input=xxx"), so I changed those buttons to use the native pan, tilt and zoom commands of the Companion module.  I realize that I've made a double change, but if together they eliminate the problem, then I'm ok with that.

BBrazner, 2022-10-27:
Please re-open this issue.  The problem has indeed recurred after re-enabling the module. Again, I'm running Companion 2.3.1 with three instances of the module connected to three BD P200 cameras all at 3418-LTS. What can I do to help you diagnose the problem?

BBrazner, 2022-11-25:
Please update this issue.  To avoid the problem, my current practice is to keep my three P200s disabled.  If I need to perform a function on one of the cameras, I enable it, perform the function, then quickly disable it.  In this way, the problem does not occur. However, one of my cameras was "stuck" today after several days of not using the module for  it.  And sure enough, it turns out the module for it was enabled (one of my assistants must have left it enabled). Rebooting the camera via the WebUI "unsticks" the camera. As mentioned in earlier posts, I'm on Companion 2.3.1 and my cameras are at 3418-LTS. I know it's easy to say that I should upgrade the cameras to NDI 5, but I think at this point I'm running what all would consider to be a  supported configuration, so I think you need to take a look at the code.  

Side note: it would be nice to have a reboot action in the module.

bryce-seifert, 2023-03-19:
If you're still experiencing this issue, please give the v3 version, now in beta, a try and let us know if it occurs in that version.

closed 2023-03-19 by bryce-seifert