bitfocus/companion-module-birddog-ptz #81: P200 issues  Companion (2.4.0+4735-beta-0700874e) + FW 5.0.0.64

joakyhsf69, 2022-10-20:
Hello, I have several P200 with firmware 5.0.0.64 and I have some problems:
-P/T Speed does not work, which has already been said in another post
-WB does not change to Manual, in the new UI it is called user, it may have something to do with it
-It is not a problem in itself, but there is no action to call the OSD Menu, it would be nice to have it
-The most important, I don't know if it's because of the version of the companion (2.4.0+4735-beta-0700874e) but I can't add new feedbacks, the menu is empty, the ones I already had configured work for me, but I can't add new ones .
For the rest, the module works quite well. I suppose that the update of birddog also has to do with it. Good job and thanks.

bryce-seifert, 2022-10-21:
> -WB does not change to Manual, in the new UI it is called user, it may have something to do with it
Yeah the API has changed to return `"WbMode":"USER"` instead of the previous `MANUAL1`, so we'll work on a fix for that.

> -The most important, I don't know if it's because of the version of the companion (2.4.0+4735-beta-0700874e) but I can't add new feedbacks, the menu is empty, 
Are there any error being returned in the Companion log tab? Are feedbacks working for other modules? IE is it just the BirdDog feedbacks not loading?

joakyhsf69, 2022-10-21:
> Yeah the API has changed to return "WbMode":"USER" instead of the previous MANUAL1, so we'll work on a fix for that.

OK, I assumed it would be something like that

> If you would like to try it now, you can use the Custom Visca Command and enter this value 8101060610FF to toggle the OSD on/off

Yes it works, that's enough

> Are there any error being returned in the Companion log tab? Are feedbacks working for other modules? IE is it just the BirdDog feedbacks not loading?

I don't know about other modules, but the internal companion feedbacks are listed

jmcc94, 2022-10-25:
@joakyhsf69 - Are you able to try the latest beta 4737.. apparently it's got a fix for a UI bug that was stopping these feedbacks from showing in the list

joakyhsf69, 2022-10-25:
Yes, beta 4737 fixes the feedback problem. Thank you very much

closed 2022-11-03 by joakyhsf69