bitfocus/companion-module-birddog-ptz #86: Variables fail to get set at Companion startup

BBrazner, 2022-10-25:
I have 2.3.1 Companion on Mac M1 Mini, with BD P200 cameras running 3418-LTS FW.

I have a trigger that runs at Companion Startup.  The actions set the initial speed values for pan, tilt and zoom (e.g., PTZ-Pan Speed: Speed Value 3). The log indicates the trigger fires at startup.  Yet, examination of the variables ( e.g., $(xxx:pan_speed) ) reveal the actions never took effect.

bryce-seifert, 2022-10-27:
I think this might be similar to https://github.com/bitfocus/companion-module-birddog-ptz/issues/78

Are all of the actions on the same trigger?
Does the BirdDog web UI reflect the correct values? Or are those unchanged as well?

BBrazner, 2022-10-27:
For this issue, I'm speaking of multiple actions all under a single trigger that runs at Companion Startup.  The trigger runs (per the Companion log), but the values in the WebUI do not change, nor do the module variables change. It's as if the actions in the trigger (9 in all) were not processed.

BBrazner, 2022-10-27:
As for #78, I should point out I'm running 3418-LTS, not NDI 5.  I have buttons for speed up and speed down for pan, tilt and zoom for 3 P200 cameras, and they take effect right away, with the exception of tilt speed down which I've documented in #85.

bryce-seifert, 2022-10-28:
Thanks for the clarification. We've reached out to BirdDog as this one is stumping us.

We confirmed that Companion is sending all the actions properly, but sometimes the camera is not updating all values properly. We're also able to replicate this outside of Companion.

We'll keep investigating as we await their response.

BBrazner, 2022-10-28:
I'm glad you have been able to confirm the problem. The camera seems to update the p/t/z speed settings at other times in the session. Do you think maybe adding a delay (e.g., 5 seconds) in the startup trigger might give the camera time to "wake up"? Or maybe a delay between each action? Or maybe, there's another command we can send before the speed commands that might wake up the camera?

bryce-seifert, 2022-10-28:
If you're not putting the camera into standby mode, the camera should be awake and ready to get commands at startup so I don't think a delay should be needed there.

In my testing, adding a relative delay between each action, even as little as 500ms, the camera seems to get the updated values, so that could be a possible workaround.

[image]

We're also looking at possibly making one action that allows you to adjust the pan, tilt, and zoom speed all with one action as another possible workaround.

BBrazner, 2022-10-31:
Thanks for letting me know about adding a 500ms delay between each action. I'll give that a try.

Side question: I have some enhancement suggestions for the module. Do I use this forum for that?  If yes, should I lump them into one post or make a post for each one?

bryce-seifert, 2022-11-01:
Yes, making a new issue on Github is the best place for suggestions.

I would suggest lumping if they're related to each other. If not, you can make separate issues so that we can address each directly.

bryce-seifert, 2023-03-19:
Merging this with related issue https://github.com/bitfocus/companion-module-birddog-ptz/issues/78

closed 2023-03-19 by bryce-seifert