bitfocus/companion-module-birddog-ptz #87: Variables in triggers do not fire

obcocav, 2022-10-29:
I’m trying to use a trigger to to change iris value when zoom is < or > a value.  The feedback itself updates fine but even when a condition is met it does not fire the trigger.  For example “check variable value” for zoom position > 15X then Iris 2.8 does not work even when companion. Correctly sees the zoom position as 20X.

obcocav, 2023-01-31:
@bryce-seifert.  FWIW.  I have been using a work around by multiplying the variable by 1.  Seems to remove the degrees symbol.  So I have a dozen or so triggers to “correct” the values that fire every 3 seconds.  It works but Sure would be nice to get rid of the clutter with native ability.  Hope this helps!  Thanks for all the work you do!

closed 2023-03-18 by bryce-seifert

bryce-seifert, 2023-03-19:
@obcocav In v3, now in beta, zoom will now be available in as a raw integer with the `zoom_position_int` variable