bitfocus/companion-module-birddog-ptz #89: Birrddog PTZ Commands dont show up in comand dropdown

TreborKutscher, 2022-12-01:
The birddog PTZ dont show up in the dropdown for comands as the internal ones do.

But the connetion to the camera is set to green/working as the ip is valid

jmcc94, 2022-12-01:
@TreborKutscher - What version of Companion are you running, and what FW is on the camera?

TreborKutscher, 2022-12-01:
Hello, i  run 2.4.0 (2.4.0+4877-v240-cd0b68c6)
Firmware of Camera:  BirdDog EYES P4K 5.0.064  MCU 10 also with SDI Board Update 

But i would assume the Problem is more an internal companion issue

[image]
Module says connected

[image]
But the action presets dont show up.

Iam used to program companion and button actions  even if non of the modules is having a connection.

ThoomZ, 2022-12-20:
Is there any progress in the resolution of this bug that also occurs on my side in version 2.4.1 of Companion
Thanks

swill97, 2023-01-23:
Can also confirm that this is an issue, running 2.4.1 of Companion, with P200's and P400 running 5.0.064.

Is there any progress with this since it has been nearly two months.

bryce-seifert, 2023-01-23:
In the module settings, are you using "Auto Detect" or selecting your camera?

Are there any messages in the Companion log, such as
`instance(BirdDog PTZ): Detected camera model: P200A4A5`

swill97, 2023-01-23:
> In the module settings, are you using "Auto Detect" or selecting your camera?
> 
> Are there any messages in the Companion log, such as `instance(BirdDog PTZ): Detected camera model: P200A4A5`

I've tried both Auto Detect and Selecting the camera, both show nothing in the log other than detected camera model and connected (including debug logs)

bryce-seifert, 2023-01-23:
Would you be able to "Export support bundle" on the log page, and send that to me via the [Slack group](https://join.slack.com/t/bitfocusio/shared_invite/zt-o1i0mohx-p_m5I7fUEUJ1BBu_KrDN2Q)?

jmcc94, 2023-01-23:
I've seen something similar with a earlier version... it was a combination with the module and a bug in the UI... Should have been sorted well before v2.4.1

swill97, 2023-01-23:
> Would you be able to "Export support bundle" on the log page, and send that to me via the [Slack group](https://join.slack.com/t/bitfocusio/shared_invite/zt-o1i0mohx-p_m5I7fUEUJ1BBu_KrDN2Q)?

I've tried signing into Slack but its throwing errors preventing me from signing in, I've got the files ready to go though.

jmcc94, 2023-01-23:
Are you able to search for the actions?

bryce-seifert, 2023-01-23:
@swill97 Feel free to email it to bryceapps@icloud.com

KimP78, 2023-02-02:
Hi all, has there been any update/resolution for this? Would love to hear the work-arounds that you have used or if you have rolled back software versions.

bryce-seifert, 2023-02-02:
@KimP78 No resolution yet. I'm having a hard time recreating the issue on my end, and not seeing anything in the logs so it's a bit of a stumper.

To confirm, what version of Companion are you on?

KimP78, 2023-02-02:
I'm running companion at v2.4.1, and I've got birddogs on v5 and on v4.5.2418

jmcc94, 2023-02-02:
@KimP78 if you search for a command, does it show up?

KimP78, 2023-02-03:
Nope, it is just like the screenshots above, there are only internal suggestion, nothing for the birddogs

jmcc94, 2023-02-04:
@KimP78, so if you start to type the Birddog instance name  or one of the commands like 'focus' in the search bar within the actions modal, nothing comes up?

bryce-seifert, 2023-03-19:
@KimP78 @swill97 @ThoomZ @TreborKutscher
Thanks for you patience on this issue.

The v3 rewrite is now available in Companion v3 beta. I believe this issue should be fixed within the rewrite process. Let me know if you're able to reproduce it in v3.

closed 2023-03-19 by bryce-seifert