bitfocus/companion-module-birddog-ptz #91: Does not retry connection with P200 after it was previously offline

schwtyl, 2023-01-12:
In our church, the StreamDeck PC generally stays on, but the AV system (including the BirdDog cameras) are turned off when not in use. However, on Sunday morning the AV system is turned on, but the StreamDeck never recognizes the cameras are back online, even after several minutes, and even after pushing buttons with actions on those cameras.  We then need to reset the Companion and/or disable/re-enable the PTZ modules, which is not sustainable across our volunteers of varying experience.

As you can see in the log, the PTZ module detects the P200 is offline within a minute or so of it being unplugged. But even minutes after being plugged back in, the module doesn't seem to make any attempts to reconnect, until I disable/re-enable the module. 

[companion_log_20230112-1615.txt](https://github.com/bitfocus/companion-module-birddog-ptz/files/10406463/companion_log_20230112-1615.txt)

BitFocus Companion: latest stable (2.4.1+4898-v2-4.1-9646192a)
BirdDog P200 firmware: latest NDI5 firmware (5.0.064)

schwtyl, 2023-01-22:
This previously worked as expected - is this perhaps a side effect of the new API interface?

bryce-seifert, 2023-03-19:
The reconnection status should now update properly in Companion v3 version of the module, now in beta.

closed 2023-03-19 by bryce-seifert

schwtyl, 2024-01-28:
@bryce-seifert I'm still observing this behavior (in stable v3.2), that the P200 connection remains offline in Companion even several minutes after the device was turned in (restarting Companion resolves this, though it was be great if this wasn't necessary ever time).

3.2.0 (3.2.0+6663-stable-b83f412f)