bitfocus/companion-module-birddog-ptz #101: Bug: Camera setup fails for P100

sphlabs, 2023-04-05:
I am having problems with my P100s, with some commands not working reliably and some variables displaying as '[object Object]'.

This probably stems from an error when performing camera setup. The following log snippet repeats constantly in the logs:
```
debug: -----Sending VISCA message: 011000050000084a81090400ff
debug: -----Sending VISCA message: 011000050000085a81090438ff
debug: -----Sending VISCA message: 011000050000085b81090462ff
debug: -----Sending VISCA message: 011000050000085c81090447ff
debug: -----Sending VISCA message: 011000050000085d81090612ff
debug: ----Camera Setup for - P100
debug: [object Object]
debug: -----Incoming VISCA message: 011100040000084a905002ff
debug: Command Error: SyntaxError: Unexpected end of JSON input
debug: -----Incoming VISCA message: 011100040000085a905003ff
debug: -----Incoming VISCA message: 011100040000085b905003ff
debug: -----Incoming VISCA message: 011100070000085c905000000000ff
debug: -----Incoming VISCA message: 0111000b0000085d90500001000200000000ff
```

Variables with the '[object Object]' value are:
Pan Speed
Tilt Speed
Zoom Speed

This is on the latest Companion v3 beta ([3.0.0 (3.0.0+5760-beta-c6884044)](https://bitfocus.io/companion/))

Camera firmware: 4.5.3790-LTS

bryce-seifert, 2023-05-29:
@sphlabs Are you still encountering this issue? I can try looking into it, but I don't have a P100 to test with.

sphlabs, 2023-05-31:
I'll have another look when I get access to the cameras again.

sphlabs, 2023-06-05:
Just tested with 3.0rc1 and still getting errors.

This is the full log from startup:
```
Starting up module class: we
Sentry disabled
Module-host accepted registration
info: Detected camera model: P100
debug: ---- Detected camera model: P100
debug: ---- Initial State for camera
info: Connected to ptz01
debug: ---- Connected to ptz01
debug: ----init udp
debug: -----Sending VISCA message: 011000050000004a81090400ff
debug: -----Sending VISCA message: 011000050000005a81090438ff
debug: -----Sending VISCA message: 011000050000005b81090462ff
debug: -----Sending VISCA message: 011000050000005c81090447ff
debug: -----Sending VISCA message: 011000050000005d81090612ff
debug: ----Camera Setup for - P100
debug: [object Object]
debug: -----Incoming VISCA message: 020100010000000001
debug: -----Incoming VISCA message: 011100040000004a905002ff
debug: -----Incoming VISCA message: 011100040000005a905003ff
debug: Unknown API variable returned from about: FirmwareVersion
debug: ---- Unknown API variable returned from about: FirmwareVersion
debug: Unknown API variable returned from about: Format
debug: ---- Unknown API variable returned from about: Format
debug: -----Incoming VISCA message: 011100040000005b905003ff
debug: Unknown API variable returned from encodesetup: ColorBitDepth
debug: ---- Unknown API variable returned from encodesetup: ColorBitDepth
debug: Unknown API variable returned from encodesetup: LoopTally
debug: ---- Unknown API variable returned from encodesetup: LoopTally
debug: Unknown API variable returned from encodesetup: ScreenSaverMode
debug: ---- Unknown API variable returned from encodesetup: ScreenSaverMode
debug: Unknown API variable returned from encodesetup: TallyMode
debug: ---- Unknown API variable returned from encodesetup: TallyMode
debug: Unknown API variable returned from encodesetup: VideoCSC
debug: ---- Unknown API variable returned from encodesetup: VideoCSC
debug: Unknown API variable returned from encodesetup: VideoSampleRate
debug: ---- Unknown API variable returned from encodesetup: VideoSampleRate
debug: Unknown API variable returned from encodesetup: ChNum
debug: ---- Unknown API variable returned from encodesetup: ChNum
debug: Unknown API variable returned from encodetransport: ndidisserv
debug: ---- Unknown API variable returned from encodetransport: ndidisserv
debug: Unknown API variable returned from encodetransport: ndidisservip
debug: ---- Unknown API variable returned from encodetransport: ndidisservip
debug: Command Error: SyntaxError: Unexpected end of JSON input
debug: Unknown API variable returned from birddogexpsetup: AeResponse
debug: ---- Unknown API variable returned from birddogexpsetup: AeResponse
debug: Unknown API variable returned from birddogexpsetup: BackLight
debug: ---- Unknown API variable returned from birddogexpsetup: BackLight
debug: Unknown API variable returned from birddogexpsetup: GainLimit
debug: ---- Unknown API variable returned from birddogexpsetup: GainLimit
debug: Unknown API variable returned from birddogexpsetup: GainPoint
debug: ---- Unknown API variable returned from birddogexpsetup: GainPoint
debug: Unknown API variable returned from birddogexpsetup: GainPointPosition
debug: ---- Unknown API variable returned from birddogexpsetup: GainPointPosition
debug: Unknown API variable returned from birddogexpsetup: HighSensitivity
debug: ---- Unknown API variable returned from birddogexpsetup: HighSensitivity
debug: Unknown API variable returned from birddogexpsetup: ShutterMaxSpeed
debug: ---- Unknown API variable returned from birddogexpsetup: ShutterMaxSpeed
debug: Unknown API variable returned from birddogexpsetup: ShutterMinSpeed
debug: ---- Unknown API variable returned from birddogexpsetup: ShutterMinSpeed
debug: Unknown API variable returned from birddogexpsetup: SlowShutterEn
debug: ---- Unknown API variable returned from birddogexpsetup: SlowShutterEn
debug: Unknown API variable returned from birddogexpsetup: SlowShutterLimit
debug: ---- Unknown API variable returned from birddogexpsetup: SlowShutterLimit
debug: Unknown API variable returned from birddogexpsetup: Spotlight
debug: ---- Unknown API variable returned from birddogexpsetup: Spotlight
debug: -----Incoming VISCA message: 011100070000005c905001060b00ff
[... 24 lines cut ...]
debug: Unknown API variable returned from birddogwbsetup: WbMode
debug: ---- Unknown API variable returned from birddogwbsetup: WbMode
debug: Unknown API variable returned from birddogpicsetup: BackLightCom
debug: ---- Unknown API variable returned from birddogpicsetup: BackLightCom
debug: Unknown API variable returned from birddogpicsetup: ChromeSuppress
debug: ---- Unknown API variable returned from birddogpicsetup: ChromeSuppress
debug: Unknown API variable returned from birddogpicsetup: HighlightComp
debug: ---- Unknown API variable returned from birddogpicsetup: HighlightComp
debug: Unknown API variable returned from birddogpicsetup: HighlightCompMask
debug: ---- Unknown API variable returned from birddogpicsetup: HighlightCompMask
debug: Unknown API variable returned from birddogpicsetup: IRCutFilter
debug: ---- Unknown API variable returned from birddogpicsetup: IRCutFilter
debug: Unknown API variable returned from birddogpicsetup: Stabilizer
debug: ---- Unknown API variable returned from birddogpicsetup: Stabilizer
debug: Unknown API variable returned from birddogpicsetup: TWODNR
debug: ---- Unknown API variable returned from birddogpicsetup: TWODNR
debug: Unknown API variable returned from birddogpicsetup: ThreeDNR
debug: ---- Unknown API variable returned from birddogpicsetup: ThreeDNR
debug: -----Incoming VISCA message: 0111000b0000005d90500f0e010e0f0f030dff
debug: -----Sending VISCA message: 011000050000004a81090400ff
debug: -----Sending VISCA message: 011000050000005a81090438ff
debug: -----Sending VISCA message: 011000050000005b81090462ff
debug: -----Sending VISCA message: 011000050000005c81090447ff
debug: -----Sending VISCA message: 011000050000005d81090612ff
```

sphlabs, 2023-07-20:
I've done a bit more digging, and have made some progress on this:

- My cameras currently have firmware version 4.5.3790-LTS installed--not v5--and some of the settings in `models.js` are currently restricted to v5; e.g. `FirmwareVersion` and `Format`.
- There is another issue that I didn't pick up from the logs before. The REST call to `birddogptzsetup` doesn't return any data, resulting in a `debug: Command Error: SyntaxError: Unexpected end of JSON input` log, which explains why none of the speed settings/feedback are working.
- I've verified that this is an issue with the camera and not this module. Using other dev tools I can see the raw response to a `birddogptzsetup` call and there is no payload.

I'll probably update the firmware to v5 next, and see how that performs (I didn't do that straight away as I thought it would be useful to fix the bugs for anyone else still using v4, but the API docs don't include any version info, which makes the job of being accurate across models more tricky).

bryce-seifert, 2023-11-08:
@sphlabs Did the update to v5 end up resolving? If so I likely need to get a camera on v4 to do more debugging, although without API version info it really is a difficult.

Let me know if you're still having this on v5

sphlabs, 2023-11-08:
That's a hard question to answer at the moment. The update to v5 broke my cameras (sync, frame drop, colour issues). They've been back and forth to Birddog three times so far and I've been waiting over a month since they were last sent in…

bryce-seifert, 2023-11-08:
Yikes… 😔