bitfocus/companion-module-birddog-ptz #110: Variables showing random readings every so often

obcocav, 2023-07-13:
Thanks again for all your work on this module!  Things have settled out and working nicely in V3 other than one issue I'm seeing.  I have a couple of P200s that seem to send random Variable values every few minutes for just a second.  Without anything changing, camera completely static, Companion receives bogus values for say Pan, tilt, zoom, position for 1 second or so and then reverts to the correct value.  This is causing triggers for exposure I have set to "randomly" fire.  Any ideas?  Thanks!

bryce-seifert, 2023-07-13:
I remember noticing this coming from the API when testing, however I haven't re-checked if this persisted on more recent firmware.

I will try and find some time to test and see if I can get enough data for BD to resolve.

bryce-seifert, 2024-01-30:
@obcocav If you want to try the latest Companion beta, I think I was able to resolve this for Pan + Tilt.

It seems like the camera is sending garbage VISCA messages every so often.
I was able to account for this easily for pan/tilt, but still trying to figure out zoom since the garbage it sends is a little different