bitfocus/companion-module-birddog-ptz #111: VISCA ON does not work after update

MichaelOlsenDK, 2023-07-13:
I used to have a Companion setup with buttons to turn the cameras (P200) ON and OFF. It was done using the VISCA codes according to the documentation for P200.
Now after a major Biddog update yesterday (MCU 26, FW 5.5.094) the VISCA ON does no longer work.
I have also tried to reset the Companion software and use the build-in funtions "Cam Standby" and "Cam On" and only "Cam Standby" works. In Companion I have tried "auto" in the deteting the camera model, but I have also tried to set it manually (A4/A5).
I simply can't turn the camera ON, after it has gone in standby mode.
It worked before the Birdog update and I have also upgraded Companion to the latest beta verison (I also tried the "release candidate" and the "experimental version"). Any idea?

bryce-seifert, 2023-07-20:
I'm able to recreate this issue, but am still working on determining a cause

MichaelOlsenDK, 2023-07-20:
It works in version 2.4.3

Sendt fra Outlook til Android<https://aka.ms/AAb9ysg>

bryce-seifert, 2023-07-20:
Thanks for confirming. Nothing has changed with this specific section of module code between the two versions, so my best guess is something happened with Nodejs and Buffers, since Companion v3 moved from 14 to 18. I'll keep digging and hopefully have a fix soon.

asukumite, 2023-12-05:
Is there any update on this? Having the same issue.

crbon, 2024-01-16:
Any update on this?
Having the same issue with a P200 on the current latest firmware (MCU 27, FW 5.5.114)

Is the issue expected to be with Companion, Companion module or the Birddog camera / VISCA?

bryce-seifert, 2024-01-16:
I have an open ticket with BirdDog that I’ve been waiting for a response on.

In my testing nothing has changed in terms of the command being sent, so I’m not sure what the error is.

crbon, 2024-01-17:
From my testing outside of the Companion environment, it seems that the VISCA IP Power On command is not working. I've reached out to Birddog support. I'll post if I have an update.

bryce-seifert, 2024-01-30:
@crbon @MichaelOlsenDK I think I might have finally cracked this one. It seems like sending Zoom inquiry VISCA messages while in standby was causing the camera to lock up and not wake. I've changed it to not send these messages while in standby, and also reported the issues to BirdDog. I'm using 5.5.114 FW too.

Can you please try the latest Companion beta and let me know if it works for you?

crbon, 2024-01-31:
@bryce-seifert unfortunately it's not working for me 🙁

Initially I thought it was a problem with the BirdDog firmware (I'm sure it's still not perfect), however, after some more testing it seems that this companion module is the issue. I should have a solution ready soon.

crbon, 2024-01-31:
Should be fully fixed now #127 

Turns out the polling was still interfering with the VISCA CAM_POWER ON command.

bryce-seifert, 2024-02-01:
Thanks! I've merged it and it should be in beta soon

closed 2024-02-01 by bryce-seifert