bitfocus/companion-module-birddog-ptz #112: Visca command analysis requested

BBrazner, 2023-07-18:
Companion is running 3.1.0 (3.1.0+6050-beta-94ccd9ee, this is necessary to pick up a vMix module fix). I have the BD Visca module controlling a BD P200. The camera is responding nicely to all button presses. But, when I'm not pressing any buttons on the Streamdeck, I see the following block of messages in the module's debug log occurring over and over again every one (1) second. It would seem the module is sending (then receiving) the same 5 commands over and over again. Is this normal? Why is it sending commands when no buttons are being pressed? Note: I don't have any automation/triggers in my Companion setup. 

debug: ----Camera Setup for - P200A4A5
debug: [object Object]
debug: -----Incoming VISCA message: 0111000400002f4a905002ff
debug: -----Incoming VISCA message: 0111000400002f5a905002ff
debug: -----Incoming VISCA message: 0111000400002f5b905003ff
debug: -----Incoming VISCA message: 0111000700002f5c905000000000ff
debug: -----Incoming VISCA message: 0111000b00002f5d90500f0e07060f0f0303ff
debug: -----Sending VISCA message: 0110000500002f4a81090400ff
debug: -----Sending VISCA message: 0110000500002f5a81090438ff
debug: -----Sending VISCA message: 0110000500002f5b81090462ff
debug: -----Sending VISCA message: 0110000500002f5c81090447ff
debug: -----Sending VISCA message: 0110000500002f5d81090612ff
debug: ----Camera Setup for - P200A4A5
debug: [object Object]

bryce-seifert, 2023-07-18:
Thanks for your question. The messages you are seeing are part of a regular poll that's sent to the camera every second to get status info that is used for feedbacks and variables. You can see the revenant parts of the code (with comments) [here](https://github.com/bitfocus/companion-module-birddog-ptz/blob/c382ff9f3a41c508c0674c558a2c4291e6af9379/index.js#L526).

Things that are beings queried over VISCA include  Standby status, auto focus mode, freeze status, and P/T/Z position. So it's not so much sending an action to the camera, as it is asking for information.

Hope this is helpful, let me know if you have additional questions

BBrazner, 2023-07-18:
Thanks for you super quick reply. I tried looking up the BD Visca IP protocol, but couldn't find anything that explained 8109 commands; I just saw 8101 commands explained. Perhaps you have a resource for all BD Visca commands you could point me to. But, even if I did find a resource for 8109 commands, I would not have known if they were normal, so still I would have needed your expert analysis. Again, thank you.

bryce-seifert, 2023-07-18:
Any time! Here's a [list of the VICA commands](https://birddogtv.zohodesk.com/portal/api/kbArticles/574085000000642627/locale/en/attachments/6r81le95f4132358f4c51bd67954c5ff79410/content?portalId=edbsnfaffdf840d1eef8c81ee4ee9dfacd8a48be54e0a2e0cd1acde19c38c2612ea9e&inline=true) from BirdDog.

closed 2023-07-18 by bryce-seifert

BBrazner, 2023-07-18:
Yes, I found this list. But, where are the 8109 commands documented?

bryce-seifert, 2023-07-18:
On the 4th page is where the "inquiry" commands start, here's any example of the standby and focus mode ones.

[image]
[image]

BBrazner, 2023-07-18:
Oh geez, I was still focused on the command column. I see it all now. Thanks again, you are the best!