bitfocus/companion-module-birddog-ptz #115: Feature Request: Exposure Compensation adjustment

GarthDB, 2023-08-27:
There is a preset to toggle or change the Exposure Compensation, but I can't seem to find how to change the Exposure Compensation level within the plugin. It would be nice to have presets for increasing and decreasing it.

[image]
[image]

There is a variable for displaying the set level, but again, I can't find a way to change it.

[image]

GarthDB, 2023-08-27:
When the action is set to on, it shows a number input for setting the value.

This is a good workaround, but having presets to change it up or down would still be nice.

[image]

closed 2023-10-24 by bryce-seifert

bryce-seifert, 2023-11-02:
@GarthDB This is now included in the Companion beta. Please give it a try, and let me know if you have any issues.