bitfocus/companion-module-birddog-ptz #124: P200 Exposure Gain feedback value and Iris feedback value not appears

joakyhsf69, 2023-10-29:
Hello, when I select the Exposure-Gain or Exposure-Iris feedback and select Value, the value I want does not appear and it does not work for me. Is it a bug? It has a solution? Thank you

bryce-seifert, 2023-10-29:
@joakyhsf69 Are you using the latest version of companion?
In the module settings, are you using Auto Detect or are you manually selecting P200 from the list?
Which value(s) are missing for you?

joakyhsf69, 2023-10-29:
Yes, I use the latest beta version of companion with the PTZ module v3.1.0. I manually select the P200 A4_A5, which is the model I have. When selecting Value, a box does not appear in which I can select what value I want for the feedback to be activated

bryce-seifert, 2023-10-29:
@joakyhsf69 Got it, I was able to find the issue. I've committed a change, and the fix will be in a future release. Thanks for reporting this.

closed 2023-10-29 by bryce-seifert