bitfocus/companion-module-birddog-ptz #131: P200 stopped working with Companion on latest firmware.

kineticscreen, 2024-03-01:
Having just updated to the January firmware (MCU 29) a lot of the functions no longer work in Companion. The feedbacks are pretty much all corrupted,
[image]
the simple directional controls and zoom no longer work - really the only thing that works is triggering and saving presets. The log is flooded with
`debug: Command failed http://192.168.10.4:8080/birddogptzsetup
` every second, and when trying to trigger something that doesn't work, like pan, the word `uodathtoh` appears.

Just using the Sony Visca integration can actually do the basics with the camera, including pan tilt zoom and preset loading and saving.

But of course it can't change the preset mode between PTZ / Camera, let alone display the status of this as a variable so it's of limited use.

bryce-seifert, 2024-03-01:
This is a confirmed bug by BirdDog with the firmware. As you found in the log, the PTZ setup API endpoint is broken in the firmware. They have not provided a timeline for fixing it.

The current workaround for the P/T/Z actions to work, should be able to select the “speed override” option and that will allow them to work

kineticscreen, 2024-03-01:
Damn. So I updated the firmware to fix the 'presets speed slowing down' bug only to end up with an even bigger one!

bryce-seifert, 2025-03-30:
Going to close this in favor of https://github.com/bitfocus/companion-module-birddog-ptz/issues/142, as it has slightly more conversation about the root of the issue

closed 2025-03-30 by bryce-seifert