bitfocus/companion-module-birddog-ptz #134: Which Firmware to use with P120?

Futo8681, 2024-05-13:
Hey, 
it seems like, there's no LTS Firmware version available for [Birddog P120](https://birddog.tv/downloads/). 
After connecting the companion module to our P120 something crashed the firmware. Since then we no longer get any video via NDI and many settings were corrupt. Also seems like there's no chance to fix the firmware, because it's not possible to enter the menu to factory reset with the IR remote. NDI encode settings are not displayed anymore in the WebUI. Only NDI decode settings. 
Any idea how to handle this? Thanks!

bryce-seifert, 2024-05-30:
The Si2 firmware should be good for the P120.

Did you get the firmware issue resolved? You should be able to update the firmware via the web UI to effectively reset the device, if not it might require some support from BirdDog.

I'm curious what version you were on when you experience the issue?

bryce-seifert, 2024-06-24:
@Futo8681 Is this resolved, or are you still having issues with this?

closed 2024-07-05 by bryce-seifert