bitfocus/companion-module-birddog-ptz #135: X1 Support

szechyjs, 2024-05-29:
I'm considering purchasing some of the new X1's and wondering if anyone has verified if they work with this module?

bryce-seifert, 2024-05-29:
There isn’t any API info on the BirdDog site for the X1, so I have not worked on integrating it. I also don’t have the X1, so I won’t be able to test it as of yet

szechyjs, 2024-05-29:
> There isn’t any API info on the BirdDog site for the X1

I just opened a support ticket for more details, will let you know what I find out.

szechyjs, 2024-06-12:
In theory this could just be accomplished using NDI, https://docs.ndi.video/docs/sdk/ndi-recv#receiver-ptz-control

bryce-seifert, 2024-06-17:
That's likely beyond the scope of this module, the goal here is support via the BirdDog API so that there can be feature parity across cameras.

szechyjs, 2024-06-24:
Response from Birddog...
> At the moment, there is only very basic API control for the cameras. However, later this year there will be a firmware update to implement API control for Q-SYS, Crestron and more.

When asked about documentation for already implemented APIs, this was their response...
> There will not be any documentation until everything is fully implemented via the update later this year.

bryce-seifert, 2024-06-24:
Well that's unfortunate. If I get access to one either physically or remotely, I should be able to see what values are currently available and build from there. But it seems like API support is still a WIP.

BBrazner, 2024-08-05:
It's August, 2024. Anyone have any idea when we can expect support for this camera?

bryce-seifert, 2024-08-05:
If you can either get BirdDog to provide API documentation or provide us a camera we can work it. But until then, there’s literally no information or way to test an implementation.

BBrazner, 2024-08-06:
Well, the generic Visca module seems to work for my purposes, so I'll stay with that for now.

WalterKThe4th, 2024-08-22:
If remote access to a camera would help you out, I just bought one and could probably facilitate that...

BirdDog also posted some [VISCA commands](https://birddogtv.zohodesk.com/portal/en/kb/articles/birddog-x1-x1-ultra-visca-commands) to their site but I'm not sure if that's what you're using under the hood to speak to the cameras or not.

It would be immensely helpful for me to have a way to trigger the AI tracking on and off via a companion button. I'm not sure how best to organize this, if I can be helpful, please let me know!

koensayr, 2024-12-05:
@bryce-seifert we should have X Series firmware now with the API (starting with v53 on X1/X1U) for example

bryce-seifert, 2024-12-06:
@koensayr Is there documentation available? I don't see the X series on the [API 2.0 docs](https://birddog.tv/AV/API/index.html#api-BasicDeviceInformation-aboutGet)

koensayr, 2024-12-06:
Sorry - you're right I didn't update the documentation, but it's exactly the same API. There are some new bits for turning on/off AI/Tracking but would be nice to just add the X series as drop downs, and then I can itterate on future PRs

bryce-seifert, 2024-12-10:
It's difficult to add as a dropdown option without documentation info for the specific models, since the module needs to know what the expected responses are and for some values what the possible ranges are.

Welcome a PR or documentation that could help get it going

bryce-seifert, 2024-12-19:
> What is the recommended way to control an X1 with companion at the moment?

I don't own an X1, and I don't have proper documentation for it, so at this point the answer unfortunately is: 🤷🏻‍♂️

MatrixTunnel, 2025-01-09:
I recently bought an X4 myself and was panicking for a moment since it wasn't compatible.. I found myself here and thanks to @jkeinath I was able to get all what I needed working but I was still without control of the AI tracking. After doing some digging I found the X1 recently had support for control added via VISCA commands. I eventually found the two commands and they work with the X4
On: `81 0A 11 54 02 FF`
Off: `81 0A 11 54 03 FF`
I know this isn't related to the API but it is a work around for those needing something in the interim

I'm willing to test things out with my X4 if it helps get things added to the module

Djwatt91, 2025-03-07:
@bryce-seifert 

birddog has now released the latest firmware unlocking API in 6.1.059.

from what i can gather here is the API documentation stated in the release notes
https://documenter.getpostman.com/view/29602224/2sAYHxn45i

if you need things tested more than happy to run either of my 2 X1's

hope this helps

bryce-seifert, 2025-03-30:
Beta support for the X1, X1 Ultra, and MAX is now available with module v3.2.0
You can install this from within the Companion v4 beta, or just install the module by downloading the .zip from the [release](https://github.com/bitfocus/companion-module-birddog-ptz/releases/tag/v3.2.0) and using the [Developer Path feature](https://youtube.com/watch?v=3PJi_XYUZkA) of Companion.

If you run into any issues, please report them here so I can try and troubleshoot.

closed 2025-03-30 by bryce-seifert