bitfocus/companion-module-birddog-ptz #141: presets not appearing in Buttons tab

alaninbelfast, 2024-09-30:
Running latest stable Companion (3.4.1, but same issue with previous build).  Birddog PTZ module loads up and connects to camera, but the folders of presets no longer appear in the 'Buttons' tab. 

[image]

bryce-seifert, 2024-09-30:
Have you tried switching the BirdDog Model to your specific camera, instead of using Auto Detect and seeing if that resolves?

alaninbelfast, 2024-10-01:
The X1 isn't specifically supported yet but advice was that other model profiles work for straightforward operations (pan/zoom etc).  But even choosing those, no presets are shown.

alaninbelfast, 2024-10-01:
Just looked at the debug - stopping and starting the connection - looks like something comes back that isn't expected ...

... with Auto Detect

debug: Only recent lines are shown here, nothing is persisted
system: ** Connection stopped **
system: ** Starting Connection from "/Applications/Companion.app/Contents/Resources/bundled-modules/birddog-ptz/main.js" **
system: ** Connection started **
Starting up module class: ye
Module-host accepted registration
error: Unrecognized camera model: X1. Using "Default" camera profile
debug: Unrecognized camera model: X1. Using "Default" camera profile
info: Connected to birddog-04AAF
debug: ---- Connected to birddog-04AAF
**debug: Camera Firmware Error: TypeError: Reduce of empty array with no initial value**

... and with P200_A2_A3

debug: Only recent lines are shown here, nothing is persisted
system: ** Connection stopped **
system: ** Starting Connection from "/Applications/Companion.app/Contents/Resources/bundled-modules/birddog-ptz/main.js" **
system: ** Connection started **
Starting up module class: ye
Module-host accepted registration
info: Connected to birddog-04AAF
debug: ---- Connected to birddog-04AAF
**debug: Camera Firmware Error: TypeError: Reduce of empty array with no initial value**

bryce-seifert, 2024-10-01:
Got it, since it's X1 related, going to close this in favor of https://github.com/bitfocus/companion-module-birddog-ptz/issues/135

It looks the X1 API has some differences between those selected models, and that's causing the issue

closed 2024-10-01 by bryce-seifert