bitfocus/companion-module-birddog-ptz #142: Bug: PTZ speed broken on BirdDog P200A2_A3 5.5.114

MattRogish, 2024-11-07:
We recently upgraded the BirdDog P200 firmware to latest (5.5.114) as well as upgrading Companion from 3.1-ish to 3.4.3. _Some_ - but not all - of our PTZ buttons in Companion 3.4.3 started panning super super slowly (like, 1 pixel per second). However, some were fine. 

Looking in the JSON for the buttons I see:

**Working Button:**
```
"options": {
	"val": "left",
	"posPan": "0000",
	"posTilt": "0000",
	"override": true,
	**"panSpeed": 11,**
	**"tiltSpeed": 9,**
	"value": 88
},
```

**Not working button:**
```
"options": {
	"val": "right",
	"posPan": "0000",
	"posTilt": "0000",
	"override": true,
	"value": 89
},
```

I think - but cannot verify - that a "working button" was a button we created at some point in the past?

The non-working button is one created with v3.4.3. I wonder if the "value" parameter is either gone or otherwise doesn't do anything? 

Also note in the BirdDog Camera Control UI the Pan Speed is a value from 0-21. In Companion, the "Speed Override" control is a value from 0-255, which seems like a mismatch. 

It's also possible it had an issue with the Module Model being set to Auto-Detect, I changed it to P200 A2_A3 but that didn't seem to make difference.

Thanks!

--
Matt

bryce-seifert, 2024-11-08:
@MattRogish I think I found the issue with the pan/tilt speed override not working, and that should be fixed for new buttons in the latest beta. Let me know if creating a new button still results in issues.

In regards to the speed values, the module should recognize the speed mode (either 0-21 or 0-255) set by the camera but it sounds like it's possibly not doing that. I will look into that more.

benjleroy, 2025-01-28:
I am seeing the same issue with the P240 Camera running 5.5.114, the current camera firmware.  

pan_speed
preset_speed
tilt_speed
zoom_speed

all showing "[object Object]"

bryce-seifert, 2025-02-07:
After investigating the PTZ speed variable issues, I think the issue stems from a bug in the 5.5.114 firmware. 

When the module polls the camera's API in 5.5.114 for the related command `/birddogptzsetup,` it receives this error:

```
ReferenceError: hardwareVersion is not defined
    at /bin/BirdDogSrvr/PTZ.js:133:16
    at Layer.handle [as handle_request] (/bin/BirdDogSrvr/node_modules/express/lib/router/layer.js:95:5)
    at next (/bin/BirdDogSrvr/node_modules/express/lib/router/route.js:137:13)
    at Route.dispatch (/bin/BirdDogSrvr/node_modules/express/lib/router/route.js:112:3)
    at Layer.handle [as handle_request] (/bin/BirdDogSrvr/node_modules/express/lib/router/layer.js:95:5)
    at /bin/BirdDogSrvr/node_modules/express/lib/router/index.js:281:22
    at Function.process_params (/bin/BirdDogSrvr/node_modules/express/lib/router/index.js:335:12)
    at next (/bin/BirdDogSrvr/node_modules/express/lib/router/index.js:275:10)
    at /bin/BirdDogSrvr/PTZ.js:20:3
    at Layer.handle [as handle_request] (/bin/BirdDogSrvr/node_modules/express/lib/router/layer.js:95:5)
```

So I believe until BirdDog fixes this error in the firmware, there is no way for the module to get the speed values for the variables, or adjust the PTZ speed

closed 2025-06-26 by bryce-seifert