bitfocus/companion-module-birddog-ptz #152: Can't Establish a Connection with X4

rheathco, 2025-04-30:
I have an X4 with updated firmware, and the 4.0 Companion software. When entering my IP address (10.1.10...) on the camera and selecting X4 from the options, I'm not getting anything. It just has the yellow spinning wheel on Companion. Advice? I can access my cameras via web UI and KBD, etc.

rheathco, 2025-04-30:
For the purpose of this test, I have disabled my firewall.

bryce-seifert, 2025-04-30:
@rheathco I had another user reach out, and they had the same issue.
I believe that the support for the API is coming to a later firmware release with the X4. I am trying to confirm that with my contacts at BirdDog currently.

fhhowdy, 2025-05-11:
I'm having similar issues with the X5Ultra.  I realize the definition for the camera isn't in the models.js, so I'm currently just studying hte code to see if I might be able to make a pull request to help out.  It would appear that birddog may have gone backwards with the firmware versions for the X5 (the latest is 0.3.0024.00).  If you'd like me to create a separate X5/X5Ultra issue, jet let me know.

rheathco, 2025-05-11:
I’ve contacted Birddog and begged them for this solution and some KBD
feature enables. No timeline — it’s really affecting my broadcast.

Ryan Heathco   |  Lead Pastor  |  Ben Davis Christian Church
(317) 241-3944  |   701 S. High School Rd. Indianapolis, IN 46241
______________________________________________________
*Equipping people for life on mission.*
*Want to know more? Go to our website: www.bdcc.org <http://www.bdcc.org>*

bryce-seifert, 2025-05-11:
> I'm having similar issues with the X5Ultra. I realize the definition for the camera isn't in the models.js, 

The reason for this is currently the X5 is not included in the API documentation, so there is no reference for features available. Once added to the API docs by BirdDog, it could easily be added

bryce-seifert, 2025-05-11:
> and some KBD feature enables. 

Current KBD APU is limited to just selecting camera, so until they add more API endpoints that’s really all it has available

fhhowdy, 2025-05-12:
I'm not impressed with BD support or the overall reliability of their products, but the image quality of their cameras is much better than the others we've tried.  Since I've got the X5Ultra (I've actually got two), I'm willing to give the modifications a shot and see how they work.  The only thing that is making me nervous is that the module doesn't appear to be doing a firmware/model cross reference.  While I might get it to work, just adding an older firmware version number to a list that contains the other models that are obviously on a higher firmware feels wrong.

rheathco, 2025-06-17:
> AFAIK the X4 is still awaiting a firmware update to be fully compatible
> with the API. So although Companion may recognize the model now, I'm not
> sure that all the endpoints are available on the camera itself
>
> What FW version does your X4 read? Are you able to see any actions or
> variables, or are those missing too?

I was provided new firmware by request, it hasn’t been released yet. But
there aren’t any presets or actions. I can now connect my device via the
Device IP option (I couldn’t before) but it doesn’t load any presets.

Ryan Heathco   |  Lead Pastor  |  Ben Davis Christian Church
(317) 241-3944  |   701 S. High School Rd. Indianapolis, IN 46241
______________________________________________________
*Equipping people for life on mission.*
*Want to know more? Go to our website: www.bdcc.org <http://www.bdcc.org>*

bryce-seifert, 2025-06-20:
@rheathco Not quite, I'm wondering what your camera is returning if you go to 

cameraip:8080/about

in your browser.

As an additional troubleshooting step, I would like to know if there are any message in the module logs (accessible via the "View logs" option for the module) when you first start the module?

MatrixTunnel, 2025-06-23:
Looks like 6.1.24 was released on 6/20/25

bryce-seifert, 2025-06-26:
Just circling back, is anyone having this issue in the latest release for the X4 Ultra, 6.1.24? If so, please let me know and provide what your camera is returning if you go to

cameraip:8080/about

in your browser.
And if there are any message in the module logs (accessible via the "View logs" option for the module) when you first start the module?

rheathco, 2025-07-20:
I’ve attempted the camera ip link, but it doesn’t reach anything.

Ryan Heathco   |  Lead Pastor  |  Ben Davis Christian Church
(317) 241-3944  |   701 S. High School Rd. Indianapolis, IN 46241
______________________________________________________
*Equipping people for life on mission.*
*Want to know more? Go to our website: www.bdcc.org <http://www.bdcc.org>*

bryce-seifert, 2025-07-20:
If the camera is not reachable by IP in a browser, it sounds like a network issue.

rheathco, 2025-07-20:
I can access via IP address for the camera, just not by cameraip:8080

Ryan Heathco   |  Lead Pastor  |  Ben Davis Christian Church
(317) 241-3944  |   701 S. High School Rd. Indianapolis, IN 46241
______________________________________________________
*Equipping people for life on mission.*
*Want to know more? Go to our website: www.bdcc.org <http://www.bdcc.org>*

bryce-seifert, 2025-07-20:
If you cannot access <your cameras up>:8080/about in the browser, something is wrong with the API endpoint either networking or firmware, and so Companion will have no way to reach the camera

rheathco, 2025-07-20:
> If you cannot access your cameras IP:8080/about in the browser, something

I can connect the camera to Companion via the IP address; however, I don’t
see options in the COMPANION UI for setting up camera controls.

Ryan Heathco   |  Lead Pastor  |  Ben Davis Christian Church
(317) 241-3944  |   701 S. High School Rd. Indianapolis, IN 46241
______________________________________________________
*Equipping people for life on mission.*
*Want to know more? Go to our website: www.bdcc.org <http://www.bdcc.org>*

bryce-seifert, 2025-07-20:
If you open the module logs, do you see any errors or log messages?