bitfocus/companion-module-birddog-ptz #153: X5 Ultra Support

fhhowdy, 2025-05-12:
I know the X5 hasn't been officially added to the API documentation, yet, but I have two of them.  I thought I'd start this issue to capture the needed info to add them to the module.  

I've gone through the API documentation here:
https://documenter.getpostman.com/view/29602224/2sAYHxn45i#intro

And captured some basic responses from the various sections using postman.

/about

```
{
    "Dns": "10.30.1.1",
    "FallbackIP": "10.30.1.58",
    "FirmwareVersion": "0.3.0024.0000",
    "Format": "birddog-cam4",
    "GateWay": "10.30.1.1",
    "HardwareVersion": "BirdDog x5 Ultra",
    "HostName": "birddog-cam4",
    "IPAddress": "10.30.1.58",
    "MCUVersion": "C0.13",
    "NetworkConfigMethod": "dhcp",
    "NetworkMask": "255.255.255.0",
    "SerialNumber": "D41234567890",
    "Status": "Active"
}
```

/version

`BirdDog X5`

/birddogptzsetup

```
{
    "FocusMode": "AUTO",
    "PanSpeed": "50",
    "PanTiltSlow": "Off",
    "Preset": "Camera",
    "PresetSpeed": "50",
    "TiltSpeed": "50",
    "ZoomSpeed": "4"
}
```

What other information would be most beneficial to facilitate the addition of the model to the module?

Thanks!!

bryce-seifert, 2025-05-24:
@fhhowdy Initial support for the X5 Ultra is now available in latest release of the module (v3.4.0, which you can download if you are using the Companion v4 beta). Note: some of the API commands are not yet implemented on the X5, so those are currently omitted in the module as well. Let me know if you run into any issues!

fhhowdy, 2025-06-17:
I wanted to follow up on this and say a HUGE Thank You!  I was under weather when you first released the change, but with your update and the latest X5 firmware, I've been able to get most of the features working.  There are a few things that the cameras are not responding to, but I need to check the API and give feedback to BirdDog for those items (PTZ preset speeds not working, PTZ Presets only work with 1-9 -- nothing higher, etc).  Those are very minor issues to work through.  Thanks again, Bryce.  I made the changes and we've successfully tested them over the past few days in production.  It's all working great!